robotology/robotology-superbuild #1446: Conda Linux CI Failure End of July 2023

traversaro, 2023-07-21:
~~~
2023-07-21T03:29:44.9175025Z [108/239] Building CXX object src/TSID/CMakeFiles/TSID.dir/src/FeasibleContactWrenchTask.cpp.o
2023-07-21T03:29:44.9175667Z [109/239] Building CXX object src/TSID/CMakeFiles/TSID.dir/src/VariableRegularizationTask.cpp.o
2023-07-21T03:29:44.9176347Z [110/239] Building CXX object src/Perception/CMakeFiles/PerceptionFeatures.dir/src/ArucoDetector.cpp.o
2023-07-21T03:29:44.9177005Z FAILED: src/Perception/CMakeFiles/PerceptionFeatures.dir/src/ArucoDetector.cpp.o 
2023-07-21T03:29:44.9182744Z /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++ -DFMT_SHARED -DPerceptionFeatures_EXPORTS -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -D_USE_MATH_DEFINES -Dcasadi_VERSION=3.6.3 -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/ParametersHandler/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/GenericContainer/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Conversions/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/System/include -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -isystem /usr/share/miniconda3/envs/test/include/eigen3 -isystem /usr/share/miniconda3/envs/test/include/opencv4 -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -O3 -DNDEBUG -std=c++17 -fPIC -MD -MT src/Perception/CMakeFiles/PerceptionFeatures.dir/src/ArucoDetector.cpp.o -MF src/Perception/CMakeFiles/PerceptionFeatures.dir/src/ArucoDetector.cpp.o.d -o src/Perception/CMakeFiles/PerceptionFeatures.dir/src/ArucoDetector.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp
2023-07-21T03:29:44.9186878Z In file included from /usr/share/miniconda3/envs/test/include/eigen3/unsupported/Eigen/CXX11/Tensor:59,
2023-07-21T03:29:44.9187549Z                  from /usr/share/miniconda3/envs/test/include/opencv4/opencv2/core/eigen.hpp:64,
2023-07-21T03:29:44.9188694Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Perception/src/ArucoDetector.cpp:20:
2023-07-21T03:29:44.9189863Z /usr/share/miniconda3/envs/test/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorMeta.h:266:91: error: 'First' was not declared in this scope; did you mean 'first'?
2023-07-21T03:29:44.9190594Z   266 |   array<Index, 1 + sizeof...(Is)> customIndices2Array(IndexType& idx, numeric_list<Index, First, Is...>) {
2023-07-21T03:29:44.9191201Z       |                                                                                           ^~~~~
2023-07-21T03:29:44.9191609Z       |                                                                                           first
2023-07-21T03:29:44.9192192Z /usr/share/miniconda3/envs/test/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorMeta.h:266:103: error: template argument 2 is invalid
2023-07-21T03:29:44.9192873Z   266 |   array<Index, 1 + sizeof...(Is)> customIndices2Array(IndexType& idx, numeric_list<Index, First, Is...>) {
2023-07-21T03:29:44.9193485Z       |                                                                                                       ^
2023-07-21T03:29:44.9194616Z /usr/share/miniconda3/envs/test/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorMeta.h: In function 'constexpr Eigen::array<Index, (1 + sizeof... (Is))> Eigen::internal::customIndices2Array(IndexType&, int)':
2023-07-21T03:29:44.9195700Z /usr/share/miniconda3/envs/test/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorMeta.h:267:37: error: 'First' was not declared in this scope; did you mean 'first'?
2023-07-21T03:29:44.9196360Z   267 |     return { static_cast<Index>(idx[First]), static_cast<Index>(idx[Is])... };
2023-07-21T03:29:44.9196794Z       |                                     ^~~~~
2023-07-21T03:29:44.9197149Z       |                                     first
2023-07-21T03:29:44.9197616Z [111/239] Building CXX object src/TSID/CMakeFiles/TSID.dir/src/QPFixedBaseTSID.cpp.o
2023-07-21T03:29:44.9198180Z [112/239] Building CXX object src/TSID/CMakeFiles/TSID.dir/src/QPTSID.cpp.o
~~~

fyi @GiulioRomualdi @S-Dafarra

traversaro, 2023-07-22:
Looking at the recent changes in the resolved environment, this seems a eigen 3.4.0  -> 3.4.1 regression.

traversaro, 2023-07-22:
> Looking at the recent changes in the resolved environment, this seems a eigen 3.4.0  -> 3.4.1 regression.

Interestingly, I could not find any 3.4.1 reference in Eigen documentation, just a branch called 3.4.1 . I wonder if that is not a release.

traversaro, 2023-07-22:
Related issues: 
* https://github.com/conda-forge/eigen-feedstock/issues/39
* https://gitlab.com/libeigen/eigen/-/issues/2702

traversaro, 2023-07-31:
Fixed by https://github.com/conda-forge/admin-requests/pull/763, see https://github.com/conda-forge/eigen-feedstock/issues/39 for more background. Anyhow, it seems that there is an incompatibility between the Eigen 3.4.1 that is going to be released at some point (see https://gitlab.com/libeigen/eigen/-/issues/2702#note_1485913208) and latest releases of OpenCV . We should report this to either Eigen or OpenCV, so that we solve it before it actually hits a release of Eigen.

closed 2023-07-31 by traversaro

jeanchristopheruel, 2023-11-08:
3.4.1 is out and unfortunately, I get OpenCV build issues. Until a fix is made, I build opencv using `-D WITH_EIGEN=OFF`.

traversaro, 2023-11-08:
> 3.4.1 is out and unfortunately, I get OpenCV build issues. Until a fix is made, I build opencv using `-D WITH_EIGEN=OFF`.

Are you sure it is out? I do not see any news in https://eigen.tuxfamily.org/index.php?title=Main_Page in or a tag in https://gitlab.com/libeigen/eigen/-/tags . As stated in https://gitlab.com/libeigen/eigen/-/issues/2702 by Eigen mantainers, the branch `3.4.1` is not a release.

jeanchristopheruel, 2023-11-08:
You're right, I'm sorry, I should have looked at the source repositories. However, as you previously mentioned, it looks like 3.4 is in fact a not-assumed [3.4.1 release](https://gitlab.com/libeigen/eigen/-/blob/3.4/Eigen/src/Core/util/Macros.h?ref_type=heads#L20). 

The issue is related to the unsupported headers of Eigen, [which is included here](https://github.com/opencv/opencv/blob/bc54f76875aa040858e5e2274acd035d9c38055d/modules/core/include/opencv2/core/eigen.hpp#L66). Concerning these headers, this is Eigen's disclaimer:

> This directory contains contributions from various users. They are provided "as is", without any support. Nevertheless,
most of them are subject to be included in Eigen in the future.

I can see that [unsupported/Eigen/CXX11/src/Tensor](https://gitlab.com/libeigen/eigen/-/tree/master/unsupported/Eigen/CXX11/src/Tensor?ref_type=heads) is in active development. 

The issue that we are facing has been fixed in Eigen Master branch [with this commit](https://gitlab.com/libeigen/eigen/-/commit/e4c40b092a209a52b882e2e46f32799582c2f430). 

Actually, I don't think we can blame Eigen for this issue since they explicitly state that these faulty headers are not supported. The blame goes to OpenCv which does not exclude this Eigen version from the build.

traversaro, 2023-11-08:
Thanks for digging the issue and providing the details! Probably it may be worth reporting this to Eigen maintainers, then they can decide if they want to backport the fix to 3.4 or not.

cantonios, 2023-11-10:
> I confirm that the Eigen branch 3.4 is in fact a [3.4.1 release](https://gitlab.com/libeigen/eigen/-/blob/3.4/Eigen/src/Core/util/Macros.h?ref_type=heads#L20) [since November 5th](https://gitlab.com/libeigen/eigen/-/commit/bae907b8f6078b1df290729eef946360315bd312).

It is not a release, it is the branch we use to prepare releases.  Releases will be announced, have a git tag, and a release archive under https://gitlab.com/libeigen/eigen/-/releases.

jeanchristopheruel, 2023-11-10:
Thank you for clarifying. I appreciate the update on the Eigen branch 3.4.1 !