robotology/robotology-superbuild #1453: Report opencv + Eigen 3.4.1 problem

traversaro, 2023-08-03:
See https://github.com/robotology/robotology-superbuild/issues/1446#issuecomment-1658633500, ideally before the Eigen 3.4.1 release .

jeanchristopheruel, 2023-11-08:
I just made a Pull Request to address the issue.
https://github.com/opencv/opencv/pull/24513

traversaro, 2023-11-08:
I reported the issue to Eigen mantainers with the details @jeanchristopheruel kindly provided in https://gitlab.com/libeigen/eigen/-/issues/2745 .

traversaro, 2023-11-09:
> I reported the issue to Eigen mantainers with the details @jeanchristopheruel kindly provided in https://gitlab.com/libeigen/eigen/-/issues/2745 .

And proposed a fix in https://gitlab.com/libeigen/eigen/-/merge_requests/1412, thanks @jeanchristopheruel !

traversaro, 2023-11-09:
As the error in itself is tracked on the Eigen bugtracker, I will close the issue here.

closed 2023-11-09 by traversaro