robotology/robotology-superbuild #1463: Problem with suberbuild on matlab

fils99, 2023-08-29:
I have found the following error using superbuild on matlab:

[image]

Any suggestions?

traversaro, 2023-08-29:
Yes. In a nutshell, it should be sufficient to run:
~~~
sudo aptinstall matlab-support
~~~
and specify your MATLAB installation directory and when there is the question "Rename MATLAB's GCC libraries?" answer "Yes".

If there are no configuration question when you run , probably that means that the package is already installed. In that case, you can just reconfigure it with the command:
~~~
sudo dpkg-reconfigure matlab-support
~~~
and again, specify your MATLAB installation directory and when there is the question "Rename MATLAB's GCC libraries?" answer "Yes".

See https://github.com/robotology/robotology-superbuild/issues/64 for more background on the problem.

closed 2023-08-29 by traversaro

traversaro, 2023-08-29:
I opened a PR to add this information in the documentation: https://github.com/robotology/robotology-superbuild/pull/1464 .