robotology/robotology-superbuild #1499: Error with YARP v3.8.1 when compiling robotology-superbuild v2023.08.0 on Ubuntu 22.04

martinaxgloria, 2023-10-12:
After the upgrade of `icub-head` OS of iCubGenova11 to Ubuntu 22.04, me and @Nicogene tried to compile the `robotology-superbuild` on tag `v2023.08.0`. We enabled the `ROBOTOLOGY_ENABLE_ICUB_HEAD` tag and disable the `ROBOTOLOGY_USES_GAZEBO` one. Then we started the building phase and we retrieved this error:

```bash
[ 23%] Linking CXX shared library ../../../lib/libYARP_sig.so
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libavcodec.a(lpc.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: failed to set dynamic section sizes: bad value
```

- Here the [CMakeCache.txt](https://github.com/robotology/robotology-superbuild/files/12882238/CMakeCache.txt).
- We compiled it from scratch, so we cloned the superbuild, installed the apt dependencies and created a new build folder. I can do it again. 
- I'm using cmake version 3.22.1, so the default one for Ubuntu 22.
- I tried to `locate /usr/lib/x86_64-linux-gnu/libavcodec.so` but search gave three results:

```
/usr/lib/x86_64-linux-gnu/libavcodec.so
/usr/lib/x86_64-linux-gnu/libavcodec.so.58
/usr/lib/x86_64-linux-gnu/libavcodec.so.58.54.100
```

cc @Nicogene @traversaro

traversaro, 2023-10-12:
> Here the [CMakeCache.txt](https://github.com/robotology/robotology-superbuild/files/12882238/CMakeCache.txt).

Sorry, in this context the one needed is the CMakeCache.txt of YARP, i.e. `build/src/YARP/CMakeCache.txt` , not the one of the superbuild.

martinaxgloria, 2023-10-12:
Ops, sorry. Here you are:
[CMakeCache.txt](https://github.com/robotology/robotology-superbuild/files/12882541/CMakeCache.txt)
@traversaro

traversaro, 2023-10-12:
From the CMakeCache.txt the problem is clear:

~~~
//FFMPEG's avcodec include directory
FFMPEG_avcodec_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's avcodec library
FFMPEG_avcodec_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavcodec.a

//FFMPEG's avdevice include directory
FFMPEG_avdevice_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's avdevice library
FFMPEG_avdevice_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavdevice.so

//FFMPEG's avfilter include directory
FFMPEG_avfilter_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's avfilter library
FFMPEG_avfilter_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavfilter.so

//FFMPEG's avformat include directory
FFMPEG_avformat_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's avformat library
FFMPEG_avformat_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavformat.so

//FFMPEG's avresample include directory
FFMPEG_avresample_INCLUDE_DIR:PATH=FFMPEG_avresample_INCLUDE_DIR-NOTFOUND

//FFMPEG's avresample library
FFMPEG_avresample_LIBRARY:FILEPATH=FFMPEG_avresample_LIBRARY-NOTFOUND

//FFMPEG's avutil include directory
FFMPEG_avutil_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's avutil library
FFMPEG_avutil_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavutil.so

//FFMPEG's swresample include directory
FFMPEG_swresample_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's swresample library
FFMPEG_swresample_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libswresample.so

//FFMPEG's swscale include directory
FFMPEG_swscale_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's swscale library
FFMPEG_swscale_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libswscale.so
~~~

As a workaround, you can just edit the CMakeCache to change `FFMPEG_avcodec_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavcodec.a` to `FFMPEG_avcodec_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavcodec.so` . However, it  would be interesting to complete delete the lines:
~~~
//FFMPEG's avcodec library
FFMPEG_avcodec_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavcodec.a
~~~

and see if the problem occurs again if we re-configure.

martinaxgloria, 2023-10-12:
Thanks for the feedback @traversaro. I tried by manually modifying the CMakeCache.txt as you suggested and then rebuild but now I have this error:

```
Consolidate compiler generated dependencies of target YARP_sig
make[5]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libavcodec.so', needed by 'lib/libYARP_sig.so.3.8.1'.  Stop.
make[4]: *** [CMakeFiles/Makefile2:4037: src/libYARP_sig/src/CMakeFiles/YARP_sig.dir/all] Error 2
make[3]: *** [Makefile:156: all] Error 2
make[2]: *** [CMakeFiles/YARP.dir/build.make:85: src/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:953: CMakeFiles/YARP.dir/all] Error 2
```

If I try to remove those lines:
```
//FFMPEG's avcodec library
FFMPEG_avcodec_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavcodec.a
```
nothing changes, they are regenerated at compilation time and the error occurs again.

martinaxgloria, 2023-10-12:
I double-checked with another Ubuntu 22.04 machine in which the last distro of robotology-superbuild is installed and this error is never occurred. Moreover, the YARP CMakeCache.txt of the other machine has linked the correct library name (FFMPEG_avcodec_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavcodec.so).

traversaro, 2023-10-12:
Are you sure that file `/usr/lib/x86_64-linux-gnu/libavcodec.so` actually exists? That would be the error you get if the file did not exists.

What happens if you run:
~~~
file /usr/lib/x86_64-linux-gnu/libavcodec.so
~~~

and 
~~~
file `readlink -f /usr/lib/x86_64-linux-gnu/libavcodec.so`
~~~

Nicogene, 2023-10-12:
This issue has been solved uninstalling and purging libavcodec-dev and reinstalling, I think we can close it thanks @traversaro !

traversaro, 2023-10-12:
Then for some reason the installation was corrupt, thanks!

traversaro, 2024-01-30:
Apparently this happened again on @rob-mau laptop, probably there is something that is messing with `libavcodec-dev` installation. If someone is experiencing the same problem, I guess:

~~~
sudo apt-get purge libavcodec-dev
sudo apt-get install libavcodec-dev
~~~

then open the `build/src/YARP/CMakeCache.txt`, and change the line:
~~~
FFMPEG_avcodec_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavcodec.a
~~~

to 

~~~
FFMPEG_avcodec_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavcodec.so
~~~

traversaro, 2024-01-30:
Actually @rob-mau problem was different:

~~~
[ 13%] Linking CXX shared library ../../../lib/libYARP_sig.so
/usr/bin/ld: /usr/local/lib/libavcodec.a(vc1dsp_mmx.o): relocation R_X86_64_PC32 against symbol `ff_pw_9' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[5]: *** [src/libYARP_sig/src/CMakeFiles/YARP_sig.dir/build.make:316: lib/libYARP_sig.so.3.9.0] Error 1
~~~

The problem there was that there was a custom ffmpeg installation in `/usr/local/lib`. To solve this problem, we had to find the source directory of this ffmpeg installation, and run `sudo make uninstall` in it. After that, starting from a clean build (and after reinstalling the packages that we uninstalled with `sudo apt-get purge libavcodec-dev`, I modified the previous comment to specify to install the dependencies with `sudo ./scripts/install_apt_dependencies.sh`) everything worked.

closed 2024-01-30 by traversaro