robotology/robotology-superbuild #1500: Do we need `include(FetchContent)` in the file `Buildbipedal-locomotion-framework.cmake`

mebbaid, 2023-10-13:
Recently I needed to upgrade my machine to Ubunut 22.04 and reinstall the superbuild. I chose to install the dependencies via apt and not the conda route. Everything was ok. Then I installed matlab on the machine and needed to enable the `ROBOTOLOGY_USES_MATLAB` option. When trying to configure via `ccmake ..`, I encountered the following issue

```
CMake Error at cmake/Fetchonnxruntimebinaries.cmake:23 (FetchContent_Declare):
   Unknown CMake command "FetchContent_Declare".
 Call Stack (most recent call first):
   cmake/Buildbipedal-locomotion-framework.cmake:71 (include)
   build/install/share/YCM/modules/FindOrBuildPackage.cmake:205 (include)
   cmake/RobotologySuperbuildLogic.cmake:93 (find_or_build_package)
   CMakeLists.txt:66 (include)
```

Notice: my `cmake --version` returns 3.22.1 and so `FetchContent_Declare` should be a defined CMake command.
What i did is to append to the top of `Buildbipedal-locomotion-framework.cmake` an `include(FetchContent)` which seems to fix the issue. 

I am not sure if this fix is necessary, or if I should look at something else.

traversaro, 2023-10-13:
It is necessary, we just need to keep it in the file were we using `FetchContent_Declare` (like IWYU), so in `Fetchonnxruntimebinaries.cmake`.

traversaro, 2023-10-31:
I may be missing something, but this should be fixed by https://github.com/robotology/robotology-superbuild/pull/1448 . Perhaps you were running an old release of the superbuild @mebbaid ?

mebbaid, 2023-10-31:
> I may be missing something, but this should be fixed by #1448 . Perhaps you were running an old release of the superbuild @mebbaid ?

It seems to be the case. Since I no longer encounter this issue after updating the superbuild. I will close this issue.

closed 2023-10-31 by mebbaid