robotology/robotology-superbuild #1503: Use proxsuite when using the superbuild

xEnVrE, 2023-10-17:
We need to use `proxsuite` for a project involving robots for which we also use the `robotology-superbuild`. Given that we were not aware of the fact that the `superbuild` fetches and installs that library, we have cloned `proxsuite` and installed on the system..

We then realized that something was wrong as our `CMake` project could not find the `proxsuite::proxsuite-vectorized` target. After some investigation we discovered that our project was indeed try to link against the `proxsuite` library installed by the `superbuild` which however is compiled with `-DBUILD_WITH_VECTORIZATION_SUPPORT:BOOL=OFF` to support `Ubuntu 20.04`.

Could you give us a suggestion for this scenario? At the moment we are forcing `CMake` to use our installation, built with vectorization, by exporting `proxsuite_DIR` differently.

Thanks

cc @traversaro @PasMarra

traversaro, 2023-10-18:
Can we have a quick chat today to understand the context on this? Thanks! @xEnVrE @PasMarra

xEnVrE, 2023-10-18:
For sure!

traversaro, 2025-05-31:
We dropped Ubuntu 20.04 support with apt dependencies in https://github.com/robotology/robotology-superbuild/pull/1802, I guess now we can just add back simde dependency and `-DBUILD_WITH_VECTORIZATION_SUPPORT:BOOL=ON` .