robotology/robotology-superbuild #1513: Enable COMPILE_ergoCubEmotions  in ergocub-software

traversaro, 2023-10-31:
@Nicogene @martinaxgloria should we enable the `COMPILE_ergoCubEmotions` CMake option in ergocub-software in robotology-superbuild? Or that CMake options is enabled/documented somewhere else in ergocub docs?

traversaro, 2023-10-31:
There is also  [`ergocub-fingers-tuning`](https://github.com/icub-tech-iit/ergocub-software/tree/master/src/tools/ergocub-fingers-tuning) but I guess that is not need for the "normal" use of the robot?

martinaxgloria, 2023-10-31:
Hi @traversaro,

if I remember well, it is not enabled by default. Moreover, there is little documentation in the [README.md](https://github.com/icub-tech-iit/ergocub-software/blob/master/src/modules/ergoCubEmotions/README.md) of the ergoCubEmotions module inside `ergocub-software`, and it is also documented [here](https://icub-tech-iit.github.io/ergocub-software/ergoCubEmotions/). Maybe we can link this documentation within `icub-tech-docs` under the ergocub sw part

mfussi66, 2023-10-31:
> There is also [`ergocub-fingers-tuning`](https://github.com/icub-tech-iit/ergocub-software/tree/master/src/tools/ergocub-fingers-tuning?rgh-link-date=2023-10-31T10%3A02%3A20Z) but I guess that is not need for the "normal" use of the robot? fyi @mfussi66

I can confirm, not used during normal operations.

traversaro, 2023-10-31:
If `ergoCubEmotions` is used for the normal operation of the robot, perhaps we can enable it by default at least in the superbuild (perhaps guarded by the `ROBOTOLOGY_ENABLE_ICUB_HEAD` option)? So users can just install the robotology-superbuild, and start using the robot, instead of having to manually do some steps (and manual steps can quickly became difficult to manage). 

To clarify, I am happy to do the modifications in the robotology-superbuild myself, I just want to make sure that affected people are aligned on this.

martinaxgloria, 2023-10-31:
> Maybe we can link this documentation within `icub-tech-docs` under the ergocub sw part

And/or we could set this flag ON by default so that the module is always installed

martinaxgloria, 2023-10-31:
> If `ergoCubEmotions` is used for the normal operation of the robot, perhaps we can enable it by default at least in the superbuild (perhaps guarded by the `ROBOTOLOGY_ENABLE_ICUB_HEAD` option)? So [...]

I agree :+1:

traversaro, 2023-10-31:
Ok, do we want to enable the option in the superbuild or in ergocub-softare ? cc @Nicogene

pattacini, 2023-10-31:
> Ok, do we want to enable the option in the superbuild or in ergocub-softare ? cc @Nicogene

I would do this upstream, in `ergocub-software`.

Nicogene, 2023-11-02:
`ergoCubEmotions` requires `YARP` and `OpenCV`, I thought to put it as OFF as default in order to minimize the required dependencies for those who need just to install the urdf of the robots.
We can make it as default but we have to change the documentation in order to install YARP and OpenCV.
On the other hand, making it ON in the superbuild would have less impact in my opinion since YARP and OpenCV are already required by other repositories

traversaro, 2023-11-24:
@Nicogene do we have any update on this decision? I am fine with both options.

pattacini, 2023-11-24:
I'm fine also with your option @Nicogene 👍🏻

Nicogene, 2023-11-24:
> @Nicogene do we have any update on this decision? I am fine with both options.

I think we can takle this next sprint alongside the installation of app xml also when `COMPILE_ergoCubEmotions` is off @martinaxgloria

martinaxgloria, 2023-11-24:
> > @Nicogene do we have any update on this decision? I am fine with both options.
> 
> I can proceed w/ a PR!

So, with this PR the flag is enabled by default and the module is installed at the superbuild level? Is it protected by the ROBOTOLOGY_ENABLE_ICUB_HEAD flag?

Nicogene, 2023-11-24:
> > > @Nicogene do we have any update on this decision? I am fine with both options.
> > 
> > 
> > I can proceed w/ a PR!
> 
> So, with this PR the `COMPILE_ergocCubEmotions` option is enabled by default and the module is installed at the superbuild level? Is it protected by the ROBOTOLOGY_ENABLE_ICUB_HEAD flag?

I personally don't have a strong opinion on this, since it is used only in ergoCub, and the option called `ICUB_HEAD` is a bit strange but we don't have and don't want to have yet another option called `ROBOTOLOGY_ENABLE_ERGOCUB_HEAD` just for this executable.
I don't see pros or cons to either option but if you prefer to make it dependent on the ROBOTOLOGY_ENABLE_ICUB_HEAD flag I change it in the PR!

pattacini, 2023-11-24:
>I don't see pros or cons to either option but if you prefer to make it dependent on the ROBOTOLOGY_ENABLE_ICUB_HEAD flag I change it in the PR!

Maybe, there are scenarios where we want to install it on a laptop for testing purposes.

martinaxgloria, 2023-11-24:
I agree with your solution @Nicogene, I was only asking for information. Thanks!

traversaro, 2023-11-24:
> > I don't see pros or cons to either option but if you prefer to make it dependent on the ROBOTOLOGY_ENABLE_ICUB_HEAD flag I change it in the PR!
> 
> Maybe, there are scenarios where we want to install it on a laptop for testing purposes so the guard would be counterproductive.

I totally agree that it is ok not to guard this on `ROBOTOLOGY_ENABLE_ICUB_HEAD`, but just for general knoledge, `ROBOTOLOGY_ENABLE_ICUB_HEAD` options should work totally fine on laptops for testing (it even works on Windows and macOS).

> I personally don't have a strong opinion on this, since it is used only in ergoCub, and the option called ICUB_HEAD is a bit strange but we don't have and don't want to have yet another option called ROBOTOLOGY_ENABLE_ERGOCUB_HEAD just for this executable.

Just fyi, I guess `ROBOTOLOGY_ENABLE_ICUB_HEAD` needs to be enabled also on ergocub-head as it compiles all the embObj devices.

Nicogene, 2023-11-24:
> Just fyi, I guess ROBOTOLOGY_ENABLE_ICUB_HEAD needs to be enabled also on ergocub-head as it compiles all the embObj devices.

Actually it doesn't because to `ergocub-head` there are attached only the realsense and lidar, that profile has to be enabled on `ergocub-torso` instead, for this reason I was not thinking about adding that option under `ROBOTOLOGY_ENABLE_ICUB_HEAD `

traversaro, 2023-11-24:
> Actually it doesn't because to `ergocub-head` there are attached only the realsense and lidar, that profile has to be enabled on `ergocub-torso` instead, for this reason I was not thinking about [...]

Ah sorry, yes I mixed up ergocub-head and ergocub-torso . Any suggestion for a better name for `ROBOTOLOGY_ENABLE_ICUB_HEAD` are welcome.

PR #1539 (2024-02-29): ergocub-software: compile ergoCubEmotions by default if OpenCV >= 4.5.2
It fixes #1513
closed 2024-02-29 by traversaro