robotology/robotology-superbuild #1519: Problem when building the superbuild at commit babf8b5

fedeceola, 2023-11-14:
Today I built the robotology-superbuild at commit `babf8b5` with the option `cmake .. -D ROBOTOLOGY_USES_PYTHON=ON -D ROBOTOLOGY_USES_GAZEBO=OFF -D ROBOTOLOGY_ENABLE_DYNAMICS=ON`. From python it's impossible to `import icub` (and indeed `icub.py` is not available in `robotology-superbuild/build/install/lib/python3/dist-packages`). With the last distro v2023.08.0 there are no problems.

traversaro, 2023-11-15:
For reference, the reference commit is https://github.com/robotology/robotology-superbuild/commit/babf8b5 . I guess this is related to the changes in https://github.com/robotology/robotology-superbuild/pull/1508 and https://github.com/robotology/robotology-superbuild/pull/1511 .

To debug the problem, can you report the content of following files:
* `robotology-superbuild/build/install/share/robotology-superbuild/setup.sh`
* `robotology-superbuild/build/src/YARP/install_manifest.txt`
* `robotology-superbuild/build/src/ICUB/CMakeCache.txt`
* `robotology-superbuild/build/src/ICUB/install_manifest.txt` 

Thanks!

traversaro, 2023-11-15:
Actually, looking at the CI log is already clear the problem, for some reason the python files are installed in the build directory instead of the install prefix:
~~~
2023-11-15T02:36:24.8751347Z -- Installing: /__w/robotology-superbuild/robotology-superbuild/build/install/bin/iKinCartesianSolver
2023-11-15T02:36:24.8754293Z -- Set runtime path of "/__w/robotology-superbuild/robotology-superbuild/build/install/bin/iKinCartesianSolver" to "$ORIGIN/:$ORIGIN/../lib:$ORIGIN/../:/__w/robotology-superbuild/robotology-superbuild/build/install/lib"
2023-11-15T02:36:24.8756997Z -- Installing: /__w/robotology-superbuild/robotology-superbuild/build/install/bin/actionsRenderingEngine
2023-11-15T02:36:24.8759812Z -- Set runtime path of "/__w/robotology-superbuild/robotology-superbuild/build/install/bin/actionsRenderingEngine" to "$ORIGIN/:$ORIGIN/../lib:$ORIGIN/../:/__w/robotology-superbuild/robotology-superbuild/build/install/lib"
2023-11-15T02:36:24.8762606Z -- Installing: /__w/robotology-superbuild/robotology-superbuild/build/src/ICUB/lib/python3/dist-packages/icub.py
2023-11-15T02:36:24.8764507Z -- Installing: /__w/robotology-superbuild/robotology-superbuild/build/src/ICUB/lib/python3/dist-packages/_icub.so
~~~
while for yarp:
~~~
2023-11-15T02:27:33.9930558Z -- Installing: /__w/robotology-superbuild/robotology-superbuild/build/install/lib/python3/dist-packages/yarp.py
2023-11-15T02:27:33.9931369Z -- Installing: /__w/robotology-superbuild/robotology-superbuild/build/install/lib/python3/dist-packages/_yarp.so
~~~

traversaro, 2023-11-15:
So the problem may be related to the code in https://github.com/robotology/icub-main/pull/820 .

traversaro, 2023-11-16:
@fedeceola can you check if https://github.com/robotology/robotology-superbuild/pull/1526 solves the issue? Thanks!

fedeceola, 2023-11-16:
Hi @traversaro, I just checked. The issue is solved. Thanks!

traversaro, 2023-11-16:
Great, thanks!

closed 2023-11-16 by traversaro