robotology/robotology-superbuild #1536: Unstable CI failure End of November 2023

traversaro, 2023-11-24:
~~~
2023-11-24T02:30:36.0548549Z [92/450] Building CXX object src/libraries/icubmod/bcbBattery/CMakeFiles/bcbBattery.dir/yarp_plugin_icubmod_bcbBattery.cpp.o
2023-11-24T02:30:36.0550512Z FAILED: src/libraries/icubmod/bcbBattery/CMakeFiles/bcbBattery.dir/yarp_plugin_icubmod_bcbBattery.cpp.o 
2023-11-24T02:30:36.0557289Z /usr/bin/c++  -DbcbBattery_EXPORTS -I/__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/bcbBattery -isystem /__w/robotology-superbuild/robotology-superbuild/build/install/include -g -fPIC   -std=c++17 -MD -MT src/libraries/icubmod/bcbBattery/CMakeFiles/bcbBattery.dir/yarp_plugin_icubmod_bcbBattery.cpp.o -MF src/libraries/icubmod/bcbBattery/CMakeFiles/bcbBattery.dir/yarp_plugin_icubmod_bcbBattery.cpp.o.d -o src/libraries/icubmod/bcbBattery/CMakeFiles/bcbBattery.dir/yarp_plugin_icubmod_bcbBattery.cpp.o -c src/libraries/icubmod/bcbBattery/yarp_plugin_icubmod_bcbBattery.cpp
2023-11-24T02:30:36.0563284Z In file included from src/libraries/icubmod/bcbBattery/yarp_plugin_icubmod_bcbBattery.cpp:10:
2023-11-24T02:30:36.0566241Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/bcbBattery/bcbBattery.h:48:77: error: 'BATTERY_OK_STANBY' is not a member of 'yarp::dev::IBattery::Battery_status'
2023-11-24T02:30:36.0568640Z    48 |     IBattery::Battery_status     battery_status = IBattery::Battery_status::BATTERY_OK_STANBY;
2023-11-24T02:30:36.0569761Z       |                                                                             ^~~~~~~~~~~~~~~~~
2023-11-24T02:30:36.0571491Z [93/450] Building CXX object src/libraries/icubmod/gazeController/CMakeFiles/gazecontrollerclient.dir/yarp_plugin_icubmod_gazecontrollerclient.cpp.o
2023-11-24T02:30:36.0573519Z [94/450] Building CXX object src/libraries/icubmod/bcbBattery/CMakeFiles/bcbBattery.dir/bcbBattery.cpp.o
2023-11-24T02:30:36.0575041Z FAILED: src/libraries/icubmod/bcbBattery/CMakeFiles/bcbBattery.dir/bcbBattery.cpp.o 
2023-11-24T02:30:36.0581196Z /usr/bin/c++  -DbcbBattery_EXPORTS -I/__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/bcbBattery -isystem /__w/robotology-superbuild/robotology-superbuild/build/install/include -g -fPIC   -std=c++17 -MD -MT src/libraries/icubmod/bcbBattery/CMakeFiles/bcbBattery.dir/bcbBattery.cpp.o -MF src/libraries/icubmod/bcbBattery/CMakeFiles/bcbBattery.dir/bcbBattery.cpp.o.d -o src/libraries/icubmod/bcbBattery/CMakeFiles/bcbBattery.dir/bcbBattery.cpp.o -c /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/bcbBattery/bcbBattery.cpp
2023-11-24T02:30:36.0587288Z In file included from /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/bcbBattery/bcbBattery.cpp:7:
2023-11-24T02:30:36.0590522Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/bcbBattery/bcbBattery.h:48:77: error: 'BATTERY_OK_STANBY' is not a member of 'yarp::dev::IBattery::Battery_status'
2023-11-24T02:30:36.0592877Z    48 |     IBattery::Battery_status     battery_status = IBattery::Battery_status::BATTERY_OK_STANBY;
2023-11-24T02:30:36.0593993Z       |                                                                             ^~~~~~~~~~~~~~~~~
2023-11-24T02:30:36.0595794Z [95/450] Building CXX object src/libraries/icubmod/parametricCalibrator/CMakeFiles/parametricCalibrator.dir/yarp_plugin_icubmod_parametricCalibrator.cpp.o
2023-11-24T02:30:36.0598489Z [96/450] Building CXX object src/libraries/icubmod/parametricCalibrator/CMakeFiles/parametricCalibrator.dir/parametricCalibrator.cpp.o
2023-11-24T02:30:36.0600785Z [97/450] Building CXX object src/libraries/icubmod/gazeController/CMakeFiles/gazecontrollerclient.dir/ClientGazeController.cpp.o
2023-11-24T02:30:36.0602288Z ninja: build stopped: subcommand failed.
~~~

This is a regression due to a change of name in https://github.com/robotology/yarp/commit/a29c6b28801e9d84f340f44a6295aeb74ef79de5 .

pattacini, 2023-11-24:
I'm gonna fix this, thanks @traversaro 👍🏻

traversaro, 2023-11-27:
The specific issue was fixed by https://github.com/robotology/icub-main/pull/915 . Now unstable is again failing, with error:
~~~
2023-11-27T15:42:15.0577189Z -- Found YCM: /__w/robotology-superbuild/robotology-superbuild/build/install/share/cmake/YCM (found version "0.15.3.6-20231117.4+git266a5a4")
2023-11-27T15:42:15.0578312Z CMake Error at /__w/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:150 (find_package):
2023-11-27T15:42:15.0578800Z   Could not find a package configuration file provided by "YARP_rosmsg" with
2023-11-27T15:42:15.0578998Z   any of the following names:
2023-11-27T15:42:15.0579011Z 
2023-11-27T15:42:15.0579187Z     YARP_rosmsgConfig.cmake
2023-11-27T15:42:15.0579481Z     yarp_rosmsg-config.cmake
2023-11-27T15:42:15.0579496Z 
2023-11-27T15:42:15.0579878Z   Add the installation prefix of "YARP_rosmsg" to CMAKE_PREFIX_PATH or set
2023-11-27T15:42:15.0580231Z   "YARP_rosmsg_DIR" to a directory containing one of the above files.  If
2023-11-27T15:42:15.0580603Z   "YARP_rosmsg" provides a separate development package or SDK, be sure it
2023-11-27T15:42:15.0580734Z   has been installed.
2023-11-27T15:42:15.0580897Z Call Stack (most recent call first):
2023-11-27T15:42:15.0581063Z   CMakeLists.txt:39 (find_package)
2023-11-27T15:42:15.0581071Z 
2023-11-27T15:42:15.0581078Z 
2023-11-27T15:42:15.0581359Z -- Configuring incomplete, errors occurred!
~~~

This is due to https://github.com/robotology/yarp/pull/3005 . We should package yarp-ros in the superbuild in case one builds with the unstable branches.

closed 2023-11-28 by traversaro