robotology/robotology-superbuild #1568: Qt platform error on yarpmanager

lindalastrico, 2024-01-09:
Hello,
We are using the latest superbuild stable release with yarp 3.8.1.
When launching yarpmanager, we face this warning:
`Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.`
Yarpmanager works fine anyway, except when trying to launch a command from the GUI:
[image]
While the Yarpmanager GUI reports "command successfully executed on... ", the command is not actually executed and there is the following error in the yarpmanager terminal:

`t.qpa.xcb: could not connect to display `
`qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.`
`This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.`

`Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.`

It looks like a Qt issue, what do you suggest?
Thank you!

traversaro, 2024-01-09:
I am not really familiar with that functionality of the clusterwidget.

Just to understand, are you able to run `yarpview`  in `icub15`  if you do not pass via the yarpmanager? Does it work if instead of running it via the "Type the command you wish to execute" you run it via as a module as part of a regular application of yarpmanager?

FYI @Nicogene or @randaz81 do you have any idea on this?

randaz81, 2024-01-09:
Not familiar with this error. Maybe some apt get install libxcb (or similar) is missing?

randaz81, 2024-01-16:
@lindalastrico any update on your issue?

lindalastrico, 2024-01-16:
@randaz81 not yet. I haven’t had the chance to access the lab pc to test it again! I will update you on Thursday, thank you!

randaz81, 2024-01-30:
Can you run a non-qt, non-yarp GUI? For example gedit, which should be based on GTK

lindalastrico, 2024-01-30:
No! On the terminal it says:
`(gedit:3836): Gtk-WARNING **: 18:01:58.624: cannot open display:`

traversaro, 2024-02-21:
To debug further the problem, can you try to launch `yarprun` on the machine that is creating problems manually, instead of launching it via the yarpmanager?

traversaro, 2025-05-31:
@lindalastrico are you still experiencing this problem, or can we close the issue? Thanks!