robotology/robotology-superbuild #1596: Propagate CMAKE_EXPORT_COMPILE_COMMANDS option?

traversaro, 2024-02-13:
Perhaps it could make sense that if `CMAKE_EXPORT_COMPILE_COMMANDS` is set to `ON` in the robotology-superbuild, we also set it to on in all the subprojects? 

@LoreMoretti

traversaro, 2024-02-14:
I saw that in https://cmake.org/cmake/help/v3.27/variable/CMAKE_EXPORT_COMPILE_COMMANDS.html that:
> This is initialized by the [CMAKE_EXPORT_COMPILE_COMMANDS](https://cmake.org/cmake/help/v3.27/envvar/CMAKE_EXPORT_COMPILE_COMMANDS.html#envvar:CMAKE_EXPORT_COMPILE_COMMANDS) environment variable, and initializes the [EXPORT_COMPILE_COMMANDS](https://cmake.org/cmake/help/v3.27/prop_tgt/EXPORT_COMPILE_COMMANDS.html#prop_tgt:EXPORT_COMPILE_COMMANDS) target property for all targets.

Can't we just tell people to set `export CMAKE_EXPORT_COMPILE_COMMANDS=1` in their `.bashrc` or in their conda environment? I know that we told people not to put stuff in their .bashrc, but this seems quite a generic think that it could make sense to have in their .bashrc. In this way, we do not need to modify our code.

traversaro, 2024-02-14:
A second point is how to ensure that the multiple `compile_commands.json` of the subproject of the superbuild can be found by VSCode. I guess colcon and catkin workspaces have exactly the same problems, and indeed I found some related code for catkin: https://gist.github.com/AlexisTM/a6bb5ee488611e15114aa190b6f79404 . Apparently colcon does this automatically, see https://colcon.readthedocs.io/en/released/user/how-to.html#cmake-packages-generating-compile-commands-json . The related code is in https://github.com/colcon/colcon-cmake/blob/1509c77ecb417dd0b47e4d54b73833fa5e4c1baa/colcon_cmake/event_handler/compile_commands.py#L87-L104 .

traversaro, 2024-02-14:
I renamed the issue to track the actual problem we want to face.

traversaro, 2024-02-14:
Other related issues in cmake tools for vs:
* https://github.com/microsoft/vscode-cmake-tools/issues/915
* https://github.com/microsoft/vscode-cmake-tools/pull/2029
* https://github.com/microsoft/vscode-cmake-tools/issues/1120

giotherobot, 2024-02-14:
> I saw that in https://cmake.org/cmake/help/v3.27/variable/CMAKE_EXPORT_COMPILE_COMMANDS.html that:
>
>
> Can't we just tell people to set `export CMAKE_EXPORT_COMPILE_COMMANDS=1` in their `.bashrc` or [...]

This does not seem to work for me: this is how i tested (vscode).

1) enable robsub conda env
2) export CMAKE_EXPORT_COMPILE_COMMANDS=1
3) open vscode from that terminal and build the superbuild
4) check if the files are generated and if the cache is correct, it is not

traversaro, 2024-02-14:
I am not sure, but I guess the environment variable is ready only once when the project is configuring for the first time. Indeed, by checking the CMake's source code this is the case: https://github.com/Kitware/CMake/blob/v3.28.3/Modules/CMakeGenericSystem.cmake#L74-L79 .

Indeed this may be quite confusing, and explicitly passing the option may be more clear.

giotherobot, 2024-02-14:
Tested the patch and it works, all the `compile_commands.json` are generated.

Then adding the following to the settings.json:
```json
    "cmake.mergedCompileCommands": "${workspaceFolder}/build/merged_compile_commands.json",
    "C_Cpp.default.compileCommands": "${workspaceFolder}/build/merged_compile_commands.json",
```
allows for intellisense to work.

traversaro, 2024-02-14:
Great, can we document this somewhere? @LoreMoretti do you have any suggestion on where to document this?

LoreMoretti, 2024-02-14:
> 
Great, can we document this somewhere? @LoreMoretti do you have any suggestion on where to document this?

What about here: https://github.com/ami-iit/ami-commons/issues/4

giotherobot, 2024-02-14:
Also in https://github.com/ami-iit/ami-commons/blob/master/doc/visual-studio-code.md maybe? Even though this is more specific to the superbuild

giotherobot, 2024-02-14:
How about shipping the settings.json file along with the superbuild repo, this would simplify the user operations needed.

traversaro, 2024-02-14:
> How about shipping the settings.json file along with the superbuild repo, this would simplify the user operations needed.

Sure. Do you think it could make sense to also enable the `CMAKE_EXPORT_COMPILE_COMMANDS` option by default? If expert users have problems with this, they can always set it to `OFF`.

giotherobot, 2024-02-14:
At this point I would say yes, so the user has to to no action to enable this features in vscode.

LoreMoretti, 2024-02-14:
For using [clangd extension](https://clangd.llvm.org/installation.html) the `merged_compile_commands.json` of [this comment](https://github.com/robotology/robotology-superbuild/issues/1596#issuecomment-1943406582) should be named `compile_commands.json`.

I have opened an issue on `clangd` for it: https://github.com/clangd/clangd/issues/1932

But for now, to be compatible with both `intellisense` and `clangd` we can just stick to the name `compile_commands.json`.