robotology/robotology-superbuild #1627: installation in {directory] failed for unknown reason.

MandooBBangBBang, 2024-03-28:
Downloading mambaforge installer 
Download of mambaforge installer completed
Installing mambaforge
Installation of mambaforge completed
Installation in C:\Users\USER\Desktop\{Directory} failed for unknown reason.
Please open an issue at https://github.com/robotology/robotology-superbuild/issues/new .

i got these error. i don't know what is going on.. somebody help me plz

traversaro, 2024-03-28:
I guess you are trying to install the matlab packages using the one line installer, right?

Can you provide us:
* MATLAB version used
* Windows version used
* The directory in which you are trying to run the installer.

MandooBBangBBang, 2024-03-31:
Thank you for your replying.
Yes i'm trying to install the matlab pkgs using the online installer. 
My matlab version is 2023b
My window version is "Window 10 Pro 22H2".
And I'm trying to install those pkgs in C:\Users\USER\Desktop\matlab_testcode

Few days ago, I thought that the reason for the installation failure was that the directory path contained non-English characters. Afterwards, the directory and folder names were changed to English, and installation is now complete.
Thank you very much!

closed 2024-03-31 by MandooBBangBBang