robotology/robotology-superbuild #1649: Error: when run tutorial (Motor Control Interfaces in Python) yarp did not import

16239014, 2024-05-14:
Installed Robotology-Superbuild from source with Conda Forge. 

Now i am trying to run an example: `https://robotology.github.io/robotology-documentation/doc/html/icub_python_basic_motor.html` But when i try to run this code of Python, i get error.

I installed the spyder (python software) in RobSub environment (in which robotology-superbuild installed) 

**ERROR LOG:** 

runfile('/home/sohail/Desktop/python folder for thesis/untitled0.py', wdir='/home/sohail/Desktop/python folder for thesis')
Traceback (most recent call last):

  File "<ipython-input-1-ab499d1c66b9>", line 1, in <module>
    runfile('/home/sohail/Desktop/python folder for thesis/untitled0.py', wdir='/home/sohail/Desktop/python folder for thesis')

  File "/usr/lib/python3/dist-packages/spyder_kernels/customize/spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "/usr/lib/python3/dist-packages/spyder_kernels/customize/spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "/home/sohail/Desktop/python folder for thesis/untitled0.py", line 2, in <module>
    import yarp

  File "/home/sohail/robotology-superbuild/build/install/lib/python3.12/site-packages/yarp.py", line 15, in <module>
    import _yarp

ImportError: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /home/sohail/robotology-superbuild/build/install/lib/python3.12/site-packages/_yarp.so)

traversaro, 2024-05-14:
It seems like for some reason spyder is not using the conda environment, and so the system `libstdc++.so.6` library is used instead of 

Can you first try if `python -c "import yarp"` works fine in a terminal in which you just call:
~~~
conda activate robsub
source /home/sohail/robotology-superbuild/build/install/share/robotology-superbuild/setup.sh
python -c "import yarp"
~~~

If that indeed works, the problem depends on spyder. 

Can you please also report the output of `conda list` in your environment? Thanks!

16239014, 2024-05-14:
> conda activate robsub
> source /home/sohail/robotology-superbuild/build/install/share/robotology-superbuild/setup.sh
> python -c "import yarp"
 **LOG:**
(base) sohail@sohail-Lenovo-V14-G3-IAP:~$ conda activate robsub
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ source /home/sohail/robotology-superbuild/build/install/share/robotology-superbuild/setup.sh
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ pyhton -c "import yarp"
pyhton: command not found
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ python -c "import yarp"
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ spyder

> Can you please also report the output of `conda list` in your environment? Thanks!

(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ conda list
# packages in environment at /home/sohail/anaconda3/envs/robsub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h69a702a_14    conda-forge
ace                       7.1.3                h59595ed_2    conda-forge
aiohttp                   3.9.3           py312h98912ed_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.11               hd590300_1    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.7.1                h59595ed_0    conda-forge
asio                      1.29.0               h59595ed_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.16               haed3651_8    conda-forge
aws-c-cal                 0.6.10               ha9bf9b1_2    conda-forge
aws-c-common              0.9.14               hd590300_0    conda-forge
aws-c-compression         0.2.18               h4466546_2    conda-forge
aws-c-event-stream        0.4.2                he635cd5_6    conda-forge
aws-c-http                0.8.1                hbfc29b2_7    conda-forge
aws-c-io                  0.14.6               h96cd748_2    conda-forge
aws-c-mqtt                0.10.3               hffff1cc_2    conda-forge
aws-c-s3                  0.5.5                h4893938_0    conda-forge
aws-c-sdkutils            0.1.15               h4466546_2    conda-forge
aws-checksums             0.1.18               h4466546_2    conda-forge
aws-crt-cpp               0.26.4               h58a74b7_3    conda-forge
aws-sdk-cpp               1.11.267             hb1af6a8_4    conda-forge
azure-core-cpp            1.11.1               h91d86a7_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h00ab1b0_1    conda-forge
azure-storage-common-cpp  12.5.0               h94269e2_4    conda-forge
bash-completion           2.11                 ha770c72_1    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hdade7a5_3    conda-forge
blosc                     1.21.5               h0f2a231_0    conda-forge
boost                     1.82.0               h8da182e_6    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet-cpp                3.25                 hfb8ada1_2    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.28.1               hd590300_0    conda-forge
c-compiler                1.7.0                hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cfitsio                   4.3.0                hbdc6101_0    conda-forge
cmake                     3.28.3               hcfe8598_0    conda-forge
compilers                 1.7.0                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.2.0           py312h8572e83_0    conda-forge
[... 366 lines cut ...]
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxinerama          1.1.5                h27087fc_0    conda-forge
xorg-libxmu               1.1.3                h7f98852_0    conda-forge
xorg-libxpm               3.5.17               hd590300_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.11               hd590300_0    conda-forge
xorg-libxt                1.3.0                hd590300_1    conda-forge
[... cut ...]
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.9.4           py312h98912ed_0    conda-forge
yarp                      0.4.0                    pypi_0    pypi
zeromq                    4.3.5                h59595ed_1    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge

traversaro, 2024-05-14:
`spyder` is not present in the environment, how did you installed it?

16239014, 2024-05-14:
> `spyder` is not present in the environment, how did you installed it?
no it is intalled in robsub version 3.8 and i also installed spyder 3.9 in base envoirment

(base) sohail@sohail-Lenovo-V14-G3-IAP:~$ conda activate robsub
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ source /home/sohail/robotology-superbuild/build/install/share/robotology-superbuild/setup.sh
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ spyder
Attribute Qt::AA_UseSoftwareOpenGL must be set before QCoreApplication is created.

traversaro, 2024-05-14:
@16239014 please also reply to my question explaining what you are copy&pasting, thanks. Furthermore, please use `~~~` code blocks, otherwise your message are quite hard to read, thanks.

16239014, 2024-05-14:
> @16239014 please also reply to my question explaining what you are copy&pasting, thanks. Furthermore, please use `~~~` code blocks, otherwise your message are quite hard to read, thanks.

~~~(base) sohail@sohail-Lenovo-V14-G3-IAP:~$ conda activate robsub
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ source /home/sohail/robotology-superbuild/build/install/share/robotology-superbuild/setup.sh
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ spyder
Attribute Qt::AA_UseSoftwareOpenGL must be set before QCoreApplication is created.
~~~

**No it is intalled in robsub version 3.8 and i also installed spyder 3.9 in base envoirment**
And i installed spyder with environment robsub activated: sudo apt install spyder

traversaro, 2024-05-14:
> No it is intalled in robsub version 3.8 and i also installed spyder 3.9 in base envoirment

How it is possible that is installed in the `robsub` environment if it is not listed in `conda list`?

> And i installed spyder with environment robsub activated: sudo apt install spyder

I would not mix conda and apt packages, as they may not be compatible one another.

16239014, 2024-05-14:
ok then guide me how to installed spyder in robsub

traversaro, 2024-05-14:
Just to clarify, `import yarp` inside robsub is working fine, so everything robotology-superbuild is working as expected, and unfortunately I do not have any experience with `spyder`. 

I guess to install `spyder` in robsub it should be sufficient to run:

~~~
conda install spyder
~~~

in a terminal in which `robsub` is active. As this may change the library versions in the environment, after you do so I suggests to recompile the superbuild.

16239014, 2024-05-14:
> I guess to install `spyder` in robsub it should be sufficient to run:

> conda install spyder
~~~
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ conda install spyder
Channels:
 - defaults
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package gst-plugins-bad-1.22.6-h06bf7ac_0 requires libxml2 >=2.11.5,<3.0.0a0, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ gst-plugins-bad is installable and it requires
│  └─ libxml2 >=2.11.5,<3.0.0a0 , which can be installed;
├─ pin-1 is installable and it requires
│  └─ python 3.12.* , which can be installed;
└─ spyder is not installable because there are no viable options
   ├─ spyder 5.5.1 would require
   │  └─ pyqtwebengine >=5.10,<5.16  but there are no viable options
   │     ├─ pyqtwebengine [5.15.10|5.15.7] would require
   │     │  └─ python >=3.10,<3.11.0a0 , which conflicts with any installable versions previously reported;
   │     ├─ pyqtwebengine [5.15.10|5.15.7] would require
   │     │  └─ python >=3.11,<3.12.0a0 , which conflicts with any installable versions previously reported;
   │     ├─ pyqtwebengine 5.15.10 would require
   │     │  └─ qt-webengine 5.15.*  but there are no viable options
   │     │     ├─ qt-webengine 5.15.9 would require
   │     │     │  └─ libxml2 >=2.10.3,<2.11.0a0 , which conflicts with any installable versions previously reported;
   │     │     └─ qt-webengine 5.15.9 would require
   │     │        └─ libxml2 >=2.9.14,<2.10.0a0 , which conflicts with any installable versions previously reported;
   │     ├─ pyqtwebengine [5.15.10|5.15.7] would require
   │     │  └─ python >=3.8,<3.9.0a0 , which conflicts with any installable versions previously reported;
   │     ├─ pyqtwebengine [5.15.10|5.15.7] would require
   │     │  └─ python >=3.9,<3.10.0a0 , which conflicts with any installable versions previously reported;
   │     └─ pyqtwebengine 5.15.7 would require
   │        └─ python >=3.7,<3.8.0a0 , which conflicts with any installable versions previously reported;
   ├─ spyder [3.2.3|3.2.4|...|4.0.1] would require
   │  └─ python >=2.7,<2.8.0a0 , which conflicts with any installable versions previously reported;
   ├─ spyder [3.2.3|3.2.4|...|3.3.1] would require
   │  └─ python >=3.5,<3.6.0a0 , which conflicts with any installable versions previously reported;
   ├─ spyder [3.2.3|3.2.4|...|5.0.5] would require
   │  └─ python >=3.6,<3.7.0a0 , which conflicts with any installable versions previously reported;
   ├─ spyder [3.3.0|3.3.1|...|5.3.3] would require
   │  └─ python >=3.7,<3.8.0a0 , which conflicts with any installable versions previously reported;
   ├─ spyder [4.0.1|4.1.0|...|5.5.1] would require
   │  └─ python >=3.8,<3.9.0a0 , which conflicts with any installable versions previously reported;
   ├─ spyder [4.2.0|4.2.1|...|5.5.1] would require
   │  └─ python >=3.9,<3.10.0a0 , which conflicts with any installable versions previously reported;
   ├─ spyder [5.1.5|5.2.2|...|5.5.1] would require
   │  └─ python >=3.10,<3.11.0a0 , which conflicts with any installable versions previously reported;
   └─ spyder [5.4.1|5.4.2|5.4.3|5.5.1] would require
      └─ python >=3.11,<3.12.0a0 , which conflicts with any installable versions previously reported.
~~~

traversaro, 2024-05-14:
@16239014 as I already stated, please do not just dump error, but explain what you are pasting. I am not an llm or chatgpt. : )

Apparently your environment is not compatible with `spyder`. Possible suggestions is to create a new environment from scratch, in which you install `spyder` since the beginning, and then re-compile the superbuild on top of the new environment.

Otherwise, you need to use spyder installed in some other way, and make spyder point to the python interpreter of the `robsub` conda environment. However, I never used spyder, so I strongly suggest to seek help on this in the spyder community, not here.

traversaro, 2024-05-28:
@16239014 I guess you went forward in some way, feel free to open a new issue if you still have problems.

closed 2024-05-28 by traversaro