robotology/robotology-superbuild #1668: error when compiling iDyntree

LoreMoretti, 2024-07-05:
When trying to compile iDyntree I get the following error:

```
[  0%] Built target iDynTree_swig_compilation
[  0%] Built target idyntree-private-fpconv
[  0%] Built target copy_matlab_model_in_build
[  0%] Built target copy_python_tests_in_build
[  9%] Built target idyntree-core
[ 11%] Built target PrivateUtilsUnitTest
[ 11%] Built target AxisUnitTest
[ 11%] Built target DirectionUnitTest
[ 12%] Built target SpatialAccUnitTest
[ 12%] Built target ArticulatedBodyInertiaUnitTest
[ 13%] Built target TwistUnitTest
[ 13%] Built target SpatialInertiaUnitTest
[ 14%] Built target WrenchUnitTest
[ 14%] Built target VectorDynSizeUnitTest
[ 14%] Built target MatrixDynSizeUnitTest
[ 15%] Built target ExpLogUnitTest
[ 15%] Built target SparseMatrixUnitTest
[ 15%] Built target EigenHelpersUnitTest
[ 16%] Built target RotationUnitTest
[ 17%] Built target EigenSparseHelpersUnitTest
[ 19%] Built target SO3UtilsUnitTest
[ 19%] Built target CubicSplineUnitTest
[ 20%] Built target SpanUnitTest
[ 21%] Built target TransformFromMatrix4x4UnitTest
[ 23%] Built target idyntree-modelio-xml
[ 24%] Built target MatrixViewUnitTest
[ 34%] Built target idyntree-model
[ 34%] Built target XMLParserUnitTest
[ 36%] Built target LinkUnitTest
[ 36%] Built target JointUnitTest
[ 36%] Built target ModelUnitTest
[ 37%] Built target ThreeAxisForceTorqueContactSensorUnitTest
[ 37%] Built target SensorsListUnitTest
[ 38%] Built target ModelTransformersUnitTest
[ 39%] Built target idyntree-solid-shapes
[ 45%] Built target idyntree-modelio
[ 46%] Built target ModelTransformersSolidShapesUnitTest
[ 46%] Built target InertialParametersSolidShapesHelpersUnitTest
[ 57%] Built target idyntree-optimalcontrol
[ 63%] Built target idyntree-visualization
[ 64%] Built target URDFModelImportUnitTest
[ 65%] Built target URDFGenericSensorImportUnitTest
[ 65%] Built target ModelExporterUnitTest
[ 66%] Built target ModelCalibrationHelperUnitTest
[ 67%] Built target PredictSensorsMeasurementUnitTest
[ 67%] Built target idyntree-high-level
[ 68%] Built target ConstraintsGroupUnitTest
[ 69%] Built target icubSensorURDFUnitTest
[ 71%] Built target IntegratorsUnitTest
[ 71%] Built target OCProblemUnitTest
[ 77%] Built target idyntree-estimation
[ 78%] Built target MultipleShootingUnitTest
[ 79%] Built target IpoptInterfaceUnitTest
[ 79%] Built target idyntree-model-view
[ 79%] Built target OptimalControlIpoptUnitTest
[ 79%] Built target L2NormUnitTest
[ 80%] Built target LinearOCOsqpUnitTest
[ 81%] Built target DynamicsLinearizationIntegrationTest
[ 82%] Built target idyntree-model-info
[ 83%] Built target VisualizerUnitTest
[ 84%] Built target KinDynComputationsUnitTest
[ 85%] Built target idyntree-model-simplify-shapes
[ 86%] Built target DynamicsIntegrationTest
[ 87%] Built target KinDynComputationsMatrixViewAndSpanUnitTest
[ 88%] Built target DenavitHartenbergIntegrationTest
[ 88%] Built target ReducedModelWithFTIntegrationTest
[ 89%] Built target iCubTorqueEstimationIntegrationTest
[ 92%] Built target idyntree-inverse-kinematics
[ 93%] Built target InertialParametersSolidShapesHelpersIntegrationTest
[ 93%] Built target BerdyHelperUnitTest
[ 95%] Built target BerdyMAPSolverUnitTest
[ 95%] Built target ExternalWrenchesEstimationUnitTest
[ 96%] Built target AttitudeEstimatorUnitTest
[ 97%] Built target SimpleLeggedOdometryUnitTest
[ 97%] Built target InverseKinematicsUnitTest
[ 97%] Built target KalmanFilterUnitTest
[ 98%] Built target InverseKinematicsMatrixViewAndSpanUnitTest
[ 98%] Built target ExtWrenchesAndJointTorquesEstimatorUnitTest
[ 99%] Built target ConvexHullHelpersUnitTest
[ 99%] Built target iDynTreeMEX
[ 99%] Building CXX object bindings/python/CMakeFiles/iDynTree.dir/iDynTreePYTHON_wrap.cxx.o
```

S-Dafarra, 2024-07-08:
Which version of SWIG do you have?

traversaro, 2024-07-08:
4.2.0 . Actually it is a fairly old conda environemnt, the problem would probably solved by updating the conda environemnt to recent version of the dependencies that in CI are working fine. I just asked to @LoreMoretti to post the error here to track it.

traversaro, 2024-09-06:
Let's close as we can't reproduce this problem, if something experience again this error please comment so we can investigate.

closed 2024-09-06 by traversaro