robotology/robotology-superbuild #1700: `ninja` in the build directory of the superbuild does not recompile all the projects

traversaro, 2024-09-02:
This is something that happened once with @LoreMoretti , but we were not able to reproduce.

xela-95, 2024-09-02:
## Repro steps
- Clone superbuild @ https://github.com/robotology/robotology-superbuild/commit/b2642de5c558d8cc903d65df1e183cba0fd823bd
- Create a conda environment following https://github.com/robotology/robotology-superbuild?tab=readme-ov-file#linux-macos-or-windows-from-source-with-dependencies-provided-by-conda-forge
- Create a `build` directory, cd inside it and configure the project with `cmake -GNinja -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX ../`
- Through `ccmake ../` set to `ON` the options `ROBOTOLOGY_ENABLE_DYNAMICS`, `ROBOTOLOGY_ENABLE_DYNAMICS_FUL` and `ROBOTOLOGY_USES_GZ` and to `OFF` `ROBOTOLOGY_USES_GZ`
- Compile with `ninja`

Then I had the necessity to use a different fork for `bipedal-locomotion-framework` and `walking-controllers`, so I cd into the corresponding repo sources (inside `src` directory of the superbuild), added the remote I needed and checkout to the needed commit.

Now, returning in the `build` directory if I run `ninja` again I got `nothing to do` and it didn't recompile. 

The only way I found to force the recompilation is cd inside the `<rob-sub-root>/build/src/repository-to-recompile` and do `ninja` and `ninja install` there.

<details><summary>conda list</summary>

```
# packages in environment at /home/acroci/mambaforge/envs/robsub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h69a702a_16    conda-forge
ace                       8.0.1                he02047a_0    conda-forge
aiohappyeyeballs          2.4.0              pyhd8ed1ab_0    conda-forge
aiohttp                   3.10.5          py312h41a817b_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.12               h4ab18f5_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.9.1                hac33072_0    conda-forge
asio                      1.29.0               h59595ed_0    conda-forge
assimp                    5.3.1                h8343317_3    conda-forge
atk-1.0                   2.38.0               h04ea711_2    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     24.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.17               he0b1f16_2    conda-forge
aws-c-cal                 0.6.11               heb1d5e4_0    conda-forge
aws-c-common              0.9.15               hd590300_0    conda-forge
aws-c-compression         0.2.18               hce8ee76_3    conda-forge
aws-c-event-stream        0.4.2                h01f5eca_8    conda-forge
aws-c-http                0.8.1               hdb68c23_10    conda-forge
aws-c-io                  0.14.7               hbfbeace_6    conda-forge
aws-c-mqtt                0.10.3               h50844eb_4    conda-forge
aws-c-s3                  0.5.7                hb7bd14b_1    conda-forge
aws-c-sdkutils            0.1.15               hce8ee76_3    conda-forge
aws-checksums             0.1.18               hce8ee76_3    conda-forge
aws-crt-cpp               0.26.6               hf567797_4    conda-forge
aws-sdk-cpp               1.11.267             hbf3e495_6    conda-forge
azure-core-cpp            1.11.1               h91d86a7_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h00ab1b0_1    conda-forge
azure-storage-common-cpp  12.5.0               h94269e2_4    conda-forge
bash-completion           2.11                 ha770c72_1    conda-forge
binutils                  2.40                 h4852527_7    conda-forge
binutils_impl_linux-64    2.40                 ha1999f0_7    conda-forge
binutils_linux-64         2.40                 hb3c18ed_1    conda-forge
blosc                     1.21.5               h0f2a231_0    conda-forge
boost                     1.84.0               h9cebb41_3    conda-forge
bullet-cpp                3.25                 hfb8ada1_2    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.33.1               heb4867d_0    conda-forge
c-compiler                1.7.0                hd590300_1    conda-forge
ca-certificates           2024.8.30            hbcca054_0    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
cfitsio                   4.4.0                hbdc6101_1    conda-forge
cli11                     2.4.1                h59595ed_0    conda-forge
cmake                     3.30.3               hf9cb763_0    conda-forge
coinhsl                   2023.11.17.0         hf90ab62_0    <unknown>
compilers                 1.7.0                ha770c72_1    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
cppzmq                    4.10.0               h2e2a08d_1    conda-forge
cudatoolkit               11.8.0              h4ba93d1_13    conda-forge
[... 345 lines cut ...]
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h4bc722e_1    conda-forge
xorg-libxinerama          1.1.5                h27087fc_0    conda-forge
xorg-libxmu               1.1.3                h4ab18f5_1    conda-forge
xorg-libxpm               3.5.17               hd590300_0    conda-forge
[... cut ...]
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.9.7           py312h66e93f0_0    conda-forge
zeromq                    4.3.5                h75354e8_4    conda-forge
zlib                      1.3.1                h4ab18f5_1    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge
zziplib                   0.13.69              he45264a_2    conda-forge
```

</details>

xela-95, 2024-09-02:
Here's also the `build.ninja` content: 
[build.ninja.txt](https://github.com/user-attachments/files/16838685/build.ninja.txt)

traversaro, 2024-09-02:
Were you able to reproduce this by following the reported step?

xela-95, 2024-09-03:
> Were you able to reproduce this by following the reported step?

I just cloned the superbuild and created a new conda environment from scratch, then i followed the steps above and reproduced it.

<details><summary>conda list</summary>

```
# packages in environment at /home/acroci/mambaforge/envs/debug_robsub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h69a702a_16    conda-forge
ace                       8.0.1                he02047a_0    conda-forge
aiohappyeyeballs          2.4.0              pyhd8ed1ab_0    conda-forge
aiohttp                   3.10.5          py310h5b4e0ec_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.12               h4ab18f5_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.9.1                hac33072_0    conda-forge
asio                      1.29.0               h59595ed_0    conda-forge
assimp                    5.3.1                h8343317_3    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               h04ea711_2    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     24.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.20               h5f1c8d9_0    conda-forge
aws-c-cal                 0.6.12               h2ba76a8_0    conda-forge
aws-c-common              0.9.17               h4ab18f5_0    conda-forge
aws-c-compression         0.2.18               h36a0aea_4    conda-forge
aws-c-event-stream        0.4.2               h161de36_10    conda-forge
aws-c-http                0.8.1               h63f54a0_13    conda-forge
aws-c-io                  0.14.8               h96d4d28_0    conda-forge
aws-c-mqtt                0.10.4               hcc7299c_2    conda-forge
aws-c-s3                  0.5.9                h10bd90f_0    conda-forge
aws-c-sdkutils            0.1.16               h36a0aea_0    conda-forge
aws-checksums             0.1.18               h36a0aea_4    conda-forge
aws-crt-cpp               0.26.8              h4f3a3cc_11    conda-forge
aws-sdk-cpp               1.11.267             h51dfee4_8    conda-forge
azure-core-cpp            1.11.1               h91d86a7_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h00ab1b0_1    conda-forge
azure-storage-common-cpp  12.5.0               h94269e2_4    conda-forge
bash-completion           2.11                 ha770c72_1    conda-forge
binutils                  2.40                 h4852527_7    conda-forge
binutils_impl_linux-64    2.40                 ha1999f0_7    conda-forge
binutils_linux-64         2.40                 hb3c18ed_1    conda-forge
blosc                     1.21.6               hef167b5_0    conda-forge
boost                     1.84.0               hb7f781d_3    conda-forge
bullet-cpp                3.25                 hcc13569_2    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.33.1               heb4867d_0    conda-forge
c-compiler                1.7.0                hd590300_1    conda-forge
ca-certificates           2024.8.30            hbcca054_0    conda-forge
cairo                     1.18.0               hbb29018_2    conda-forge
cfitsio                   4.4.0                hbdc6101_1    conda-forge
cli11                     2.4.1                h59595ed_0    conda-forge
cmake                     3.30.3               hf9cb763_0    conda-forge
compilers                 1.7.0                ha770c72_1    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
cppzmq                    4.10.0               h2e2a08d_1    conda-forge
cudatoolkit               11.8.0              h4ba93d1_13    conda-forge
cxx-compiler              1.7.0                h00ab1b0_1    conda-forge
dartsim                   6.13.2               h89be5ba_2    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
double-conversion         3.3.0                h59595ed_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
elfutils                  0.191                h924a536_0    conda-forge
expat                     2.6.2                h59595ed_0    conda-forge
fcl                       0.7.0                hadc09e8_4    conda-forge
[... 335 lines cut ...]
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h4bc722e_1    conda-forge
xorg-libxinerama          1.1.5                h27087fc_0    conda-forge
xorg-libxmu               1.1.3                h4ab18f5_1    conda-forge
xorg-libxpm               3.5.17               hd590300_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.11               hd590300_0    conda-forge
xorg-libxt                1.3.0                hd590300_1    conda-forge
xorg-libxxf86vm           1.1.5                h4bc722e_1    conda-forge
[... cut ...]
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.9.7           py310ha75aee5_0    conda-forge
zeromq                    4.3.5                h75354e8_4    conda-forge
zlib                      1.3.1                h4ab18f5_1    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge
zziplib                   0.13.69              he45264a_2    conda-forge
```

</details>

traversaro, 2024-09-03:
Great, can you provide me the exact commands you used to produce the conda environment? https://github.com/robotology/robotology-superbuild?tab=readme-ov-file#linux-macos-or-windows-from-source-with-dependencies-provided-by-conda-forge contains a lot of commands depending on the options enabled.

xela-95, 2024-09-03:
> Great, can you provide me the exact commands you used to produce the conda environment? https://github.com/robotology/robotology-superbuild?tab=readme-ov-file#linux-macos-or-windows-from-source-with-dependencies-provided-by-conda-forge contains a lot of commands depending on the options enabled.

Sure, here it is:

```console
conda create -n debug_robsub
```

```console
conda activate debug_robsub
```

```console
conda install -c conda-forge ace asio assimp boost cli11 eigen freetype glew glfw glm graphviz gsl ipopt irrlicht libjpeg-turbo libmatio libode libxml2 nlohmann_json qhull "pcl>=1.11.1" "libopencv>=4.10.0" opencv portaudio qt-main sdl sdl2 sqlite tinyxml tinyxml2 spdlog lua soxr qhull cmake compilers make ninja pkg-config tomlplusplus libzlib "ffmpeg==6.*" onnxruntime-cpp bash-completion freeglut libdc1394 libi2c libselinux-cos7-x86_64 xorg-libxau libxcb xorg-libxdamage xorg-libxext xorg-libxfixes xorg-libxxf86vm xorg-libxrandr mesa-libgl-cos7-x86_64 mesa-libgl-devel-cos7-x86_64 libxshmfence-cos7-x86_64 libxshmfence-devel-cos7-x86_64 gz-sim8
```

traversaro, 2024-09-05:
I reproduced and fixed the issue in https://github.com/robotology/ycm-cmake-modules/pull/462 . To be honest, I have no idea how this ever worked at all.

traversaro, 2024-09-05:
@xela-95 can you test after https://github.com/robotology/ycm-cmake-modules/pull/462 ? Thanks!

closed 2024-09-06 by traversaro