robotology/robotology-superbuild #1703: Submodule cmake are not initialized

FabioBergonti, 2024-09-05:
I'm installing the superbuild with dependencies configured via Conda on a Linux machine. I have enabled `ROBOTOLOGY_ENABLE_DYNAMICS`, `ROBOTOLOGY_USES_GAZEBO`, `ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS`, `ROBOTOLOGY_ENABLE_ICUB_HEAD`, `ROBOTOLOGY_USES_PYTHON`, and `ROBOTOLOGY_USES_PCL_AND_VTK` but I got this error:

```
[ 63%] Performing configure step for 'proxsuite'
Not searching for unused variables given on the command line.
loading initial cache file /home/icub/code/robotology-superbuild_conda/build/src/proxsuite/CMakeFiles/YCMTmp/proxsuite-cache-Release.cmake
CMake Error at CMakeLists.txt:29 (message):
  

  Please run the following command first:

  git submodule update --init

-- Configuring incomplete, errors occurred!
make[2]: *** [CMakeFiles/proxsuite.dir/build.make:93: src/proxsuite/CMakeFiles/YCMStamp/proxsuite-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:2956: CMakeFiles/proxsuite.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```

<details><summary>Conda Info</summary>
<p>

```
(robsub) icub@IITICUBLAP297:~/code/robotology-superbuild_conda/build/src/proxsuite(master)$ conda info

     active environment : robsub
    active env location : /home/icub/miniforge3/envs/robsub
            shell level : 1
       user config file : /home/icub/.condarc
 populated config files : /home/icub/miniforge3/.condarc
                          /home/icub/.condarc
          conda version : 24.3.0
    conda-build version : not installed
         python version : 3.10.14.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=skylake
                          __conda=24.3.0=0
                          __cuda=12.2=0
                          __glibc=2.35=0
                          __linux=6.8.0=0
                          __unix=0=0
       base environment : /home/icub/miniforge3  (writable)
      conda av data dir : /home/icub/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/icub/miniforge3/pkgs
                          /home/icub/.conda/pkgs
       envs directories : /home/icub/miniforge3/envs
                          /home/icub/.conda/envs
               platform : linux-64
             user-agent : conda/24.3.0 requests/2.31.0 CPython/3.10.14 Linux/6.8.0-40-generic ubuntu/22.04.4 glibc/2.35 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

</p>
</details> 

<details><summary>Conda List</summary>
<p>

```
(robsub) icub@IITICUBLAP297:~/code/robotology-superbuild_conda/build/src/proxsuite(master)$ conda list
# packages in environment at /home/icub/miniforge3/envs/robsub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
[... 454 lines cut ...]
xorg-libxmu               1.1.3                h4ab18f5_1    conda-forge
xorg-libxpm               3.5.17               hd590300_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.11               hd590300_0    conda-forge
xorg-libxt                1.3.0                hd590300_1    conda-forge
xorg-libxtst              1.2.5                h4bc722e_0    conda-forge
xorg-libxxf86vm           1.1.5                h4bc722e_1    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-recordproto          1.14.2            h7f98852_1002    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-util-macros          1.19.3               h7f98852_0    conda-forge
xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.9.8           py312h66e93f0_1    conda-forge
zeromq                    4.3.5                ha4adb4c_5    conda-forge
zlib                      1.3.1                h4ab18f5_1    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge
zziplib                   0.13.69              he45264a_2    conda-forge
```

</p>
</details>

traversaro, 2024-09-05:
Thanks @FabioBergonti ! I have no idea why this is happening for your and is not occurring in CI, let me check it out a bit. Is this a clean build/repo download or an existing build/repo after an update?

traversaro, 2024-09-05:
Which superbuild version are you using?

FabioBergonti, 2024-09-05:
It's a clean build/repo download on latest master

traversaro, 2024-09-05:
Which distribution are you using? Which git version? WSL2 or native Linux?

FabioBergonti, 2024-09-05:
A native Linux with distribution 22.04.4 lts
git version 2.34.1

traversaro, 2024-09-05:
Can you run the build with `make VERBOSE=1` to get the full command executed? Thanks!

traversaro, 2024-09-05:
Yes, but can you provide the full output of `make VERBOSE=1` ? Thanks!

traversaro, 2024-09-05:
Even better, can you do:

~~~
cd /home/icub/code/robotology-superbuild_conda/build
rm -rf /home/icub/code/robotology-superbuild_conda/build/src/proxsuite
rm -rf /home/icub/code/robotology-superbuild_conda/src/proxsuite
make VERBOSE=1 proxsuite
~~~

This should delete the proxsuite-related folders, and re-download again.

For reference, this is the output of this command for me (noticed that the submodules are downloaded):

~~~
(robsub) traversaro@IITBMP014LW012:~/tstfabio/robotology-superbuild/build$ make VERBOSE=1 proxsuite
/home/traversaro/miniforge3/envs/robsub/bin/cmake -S/home/traversaro/tstfabio/robotology-superbuild -B/home/traversaro/tstfabio/robotology-superbuild/build --check-build-system CMakeFiles/Makefile.cmake 0
Re-run cmake, missing byproduct: src/proxsuite/CMakeFiles/YCMTmp/proxsuite-gitsafeclone.cmake
-- Found Python3: /home/traversaro/miniforge3/envs/robsub/bin/python3.12 (found version "3.12.5") found components: Interpreter Development Development.Module Development.Embed
-- YCM not found. Bootstrapping it.
-- Fetching YCM.
CMake Warning (dev) at /home/traversaro/miniforge3/envs/robsub/share/cmake-3.30/Modules/FetchContent.cmake:1953 (message):
  Calling FetchContent_Populate(YCM) is deprecated, call
  FetchContent_MakeAvailable(YCM) instead.  Policy CMP0169 can be set to OLD
  to allow FetchContent_Populate(YCM) to be called directly for now, but the
  ability to call it with declared details will be removed completely in a
  future version.
Call Stack (most recent call first):
  cmake/YCMBootstrapFetch.cmake:103 (FetchContent_Populate)
  CMakeLists.txt:60 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Package YARP not found. Will be downloaded and built.
-- Package yarp-devices-ros not found. Will be downloaded and built.
-- Package ICUB not found. Will be downloaded and built.
-- Package ICUBcontrib not found. Will be downloaded and built.
-- Package icub-models not found. Will be downloaded and built.
-- Package ergocub-software not found. Will be downloaded and built.
-- Package robots-configuration not found. Will be downloaded and built.
-- Package GazeboYARPPlugins not found. Will be downloaded and built.
-- Package iDynTree not found. Will be downloaded and built.
-- Package idyntree-yarp-tools not found. Will be downloaded and built.
-- Package qpOASES not found. Will be downloaded and built.
-- Package BlockFactory not found. Will be downloaded and built.
-- Package WBToolbox not found. Will be downloaded and built.
-- Package UnicyclePlanner not found. Will be downloaded and built.
-- Package whole-body-estimators not found. Will be downloaded and built.
-- Package whole-body-controllers not found. Will be downloaded and built.
-- Package matioCpp not found. Will be downloaded and built.
-- Package robometry not found. Will be downloaded and built.
-- Package BayesFilters not found. Will be downloaded and built.
-- Package bipedal-locomotion-framework not found. Will be downloaded and built.
-- Package yarp-device-keyboard-joypad not found. Will be downloaded and built.
-- Package walking-controllers not found. Will be downloaded and built.
-- Found Python3: /home/traversaro/miniforge3/envs/robsub/bin/python3.12 (found version "3.12.5") found components: Interpreter
-- Package robot-log-visualizer not found. Will be downloaded and built.
-- Package resolve-robotics-uri-py not found. Will be downloaded and built.
-- Package icub-firmware-build not found. Will be downloaded and built.
-- Package yarp-device-xsensmt not found. Will be downloaded and built.
-- ROBOTOLOGY_SUPERBUILD_PYTHON_INSTALL_DIR_SETUP_SH: lib/python3.12/site-packages
-- The following RUNTIME packages have been found:

 * Git

-- The following OPTIONAL packages have been found:

 * OpenCV, Open source computer vision library, <http://opencv.org/>

-- The following REQUIRED packages have been found:

 * SWIG
[... 258 lines cut ...]
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/include/proxsuite/proxqp/utils/prints.hpp
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/include/proxsuite/proxqp/utils/random_qp_problems.hpp
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/include/proxsuite/serialization/archive.hpp
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/include/proxsuite/serialization/eigen.hpp
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/include/proxsuite/serialization/model.hpp
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/include/proxsuite/serialization/results.hpp
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/include/proxsuite/serialization/ruiz.hpp
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/include/proxsuite/serialization/settings.hpp
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/include/proxsuite/serialization/workspace.hpp
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/include/proxsuite/serialization/wrapper.hpp
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/lib/cmake/proxsuite/proxsuiteConfig.cmake
-- Installing: /home/traversaro/tstfabio/robotology-superbuild/build/install/lib/cmake/proxsuite/proxsuiteConfigVersion.cmake
[... cut ...]
/home/traversaro/miniforge3/envs/robsub/bin/cmake -E touch /home/traversaro/tstfabio/robotology-superbuild/build/src/proxsuite/CMakeFiles/YCMStamp/proxsuite-done
make[3]: Leaving directory '/home/traversaro/tstfabio/robotology-superbuild/build'
[100%] Built target proxsuite
make[2]: Leaving directory '/home/traversaro/tstfabio/robotology-superbuild/build'
/home/traversaro/miniforge3/envs/robsub/bin/cmake -E cmake_progress_start /home/traversaro/tstfabio/robotology-superbuild/build/CMakeFiles 0
make[1]: Leaving directory '/home/traversaro/tstfabio/robotology-superbuild/build'
~~~

FabioBergonti, 2024-09-05:
Deleting the proxsuite-related folders and re-download again fixed the problem of proxsuite. I can rerun the superbuild to see if now I can complete it

<details><summary>output</summary>
<p>

```
(robsub) icub@IITICUBLAP297:~/code/robotology-superbuild_conda/build(master)$ make VERBOSE=1 proxsuite
/home/icub/miniforge3/envs/robsub/bin/cmake -S/home/icub/code/robotology-superbuild_conda -B/home/icub/code/robotology-superbuild_conda/build --check-build-system CMakeFiles/Makefile.cmake 0
Re-run cmake, missing byproduct: src/proxsuite/CMakeFiles/YCMTmp/proxsuite-gitsafeclone.cmake
-- Found Python3: /home/icub/miniforge3/envs/robsub/bin/python3.12 (found version "3.12.5") found components: Interpreter Development Development.Module Development.Embed
-- YCM not found. Bootstrapping it.
-- Fetching YCM.
CMake Warning (dev) at /home/icub/miniforge3/envs/robsub/share/cmake-3.30/Modules/FetchContent.cmake:1953 (message):
  Calling FetchContent_Populate(YCM) is deprecated, call
  FetchContent_MakeAvailable(YCM) instead.  Policy CMP0169 can be set to OLD
  to allow FetchContent_Populate(YCM) to be called directly for now, but the
  ability to call it with declared details will be removed completely in a
  future version.
Call Stack (most recent call first):
  cmake/YCMBootstrapFetch.cmake:103 (FetchContent_Populate)
  CMakeLists.txt:60 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Package YARP not found. Will be downloaded and built.
-- Package yarp-devices-ros not found. Will be downloaded and built.
-- Package ICUB not found. Will be downloaded and built.
-- Package ICUBcontrib not found. Will be downloaded and built.
-- Package icub-models not found. Will be downloaded and built.
-- Package ergocub-software not found. Will be downloaded and built.
-- Package robots-configuration not found. Will be downloaded and built.
-- Package GazeboYARPPlugins not found. Will be downloaded and built.
-- Package iDynTree not found. Will be downloaded and built.
-- Package idyntree-yarp-tools not found. Will be downloaded and built.
-- Package qpOASES not found. Will be downloaded and built.
-- Package BlockFactory not found. Will be downloaded and built.
-- Package WBToolbox not found. Will be downloaded and built.
-- Package UnicyclePlanner not found. Will be downloaded and built.
-- Package whole-body-estimators not found. Will be downloaded and built.
-- Package whole-body-controllers not found. Will be downloaded and built.
-- Package matioCpp not found. Will be downloaded and built.
-- Package robometry not found. Will be downloaded and built.
-- Package BayesFilters not found. Will be downloaded and built.
-- Package bipedal-locomotion-framework not found. Will be downloaded and built.
-- Package yarp-device-keyboard-joypad not found. Will be downloaded and built.
-- Package walking-controllers not found. Will be downloaded and built.
-- Found Python3: /home/icub/miniforge3/envs/robsub/bin/python3.12 (found version "3.12.5") found components: Interpreter
-- Package robot-log-visualizer not found. Will be downloaded and built.
-- Package resolve-robotics-uri-py not found. Will be downloaded and built.
-- Package icub-firmware-build not found. Will be downloaded and built.
-- Package yarp-device-xsensmt not found. Will be downloaded and built.
-- ROBOTOLOGY_SUPERBUILD_PYTHON_INSTALL_DIR_SETUP_SH: lib/python3.12/site-packages
-- The following RUNTIME packages have been found:

 * Git

-- The following OPTIONAL packages have been found:

 * OpenCV, Open source computer vision library, <http://opencv.org/>

-- The following REQUIRED packages have been found:

 * SWIG
 * Python3

-- The following RECOMMENDED packages have not been found:

 * YCM (required version >= 0.14.0)
   Used by the build system

-- Configuring done (1.7s)
[... 251 lines cut ...]
-- Installing: /home/icub/code/robotology-superbuild_conda/build/install/include/proxsuite/serialization/archive.hpp
-- Installing: /home/icub/code/robotology-superbuild_conda/build/install/include/proxsuite/serialization/eigen.hpp
-- Installing: /home/icub/code/robotology-superbuild_conda/build/install/include/proxsuite/serialization/model.hpp
-- Installing: /home/icub/code/robotology-superbuild_conda/build/install/include/proxsuite/serialization/results.hpp
-- Installing: /home/icub/code/robotology-superbuild_conda/build/install/include/proxsuite/serialization/ruiz.hpp
-- Installing: /home/icub/code/robotology-superbuild_conda/build/install/include/proxsuite/serialization/settings.hpp
-- Installing: /home/icub/code/robotology-superbuild_conda/build/install/include/proxsuite/serialization/workspace.hpp
-- Installing: /home/icub/code/robotology-superbuild_conda/build/install/include/proxsuite/serialization/wrapper.hpp
-- Installing: /home/icub/code/robotology-superbuild_conda/build/install/lib/cmake/proxsuite/proxsuiteConfig.cmake
-- Installing: /home/icub/code/robotology-superbuild_conda/build/install/lib/cmake/proxsuite/proxsuiteConfigVersion.cmake
-- Installing: /home/icub/code/robotology-superbuild_conda/build/install/lib/cmake/proxsuite/proxsuiteTargets.cmake
make[4]: Leaving directory '/home/icub/code/robotology-superbuild_conda/build/src/proxsuite'
[100%] Completed 'proxsuite'
/home/icub/miniforge3/envs/robsub/bin/cmake -E make_directory /home/icub/code/robotology-superbuild_conda/build/CMakeFiles
[... cut ...]
/home/icub/miniforge3/envs/robsub/bin/cmake -E touch /home/icub/code/robotology-superbuild_conda/build/src/proxsuite/CMakeFiles/YCMStamp/proxsuite-done
make[3]: Leaving directory '/home/icub/code/robotology-superbuild_conda/build'
[100%] Built target proxsuite
make[2]: Leaving directory '/home/icub/code/robotology-superbuild_conda/build'
/home/icub/miniforge3/envs/robsub/bin/cmake -E cmake_progress_start /home/icub/code/robotology-superbuild_conda/build/CMakeFiles 0
make[1]: Leaving directory '/home/icub/code/robotology-superbuild_conda/build'
```

</p>
</details>

FabioBergonti, 2024-09-06:
Hi @traversaro 
The problem seems resolved
Should I close the issue?

traversaro, 2024-09-06:
Sure. I am curious how this happened in the first place, but I guess there may be some corner case not correctly handled (network failure after the repo have been cloned but while it is cloning the submodules?), that it is probably safe to ignore.

closed 2024-09-06 by traversaro