robotology/robotology-superbuild #1762: Error after installing python bindings

MissingSignal, 2024-12-09:
Dear mantainers,

When compiling the Superbuild with `ROBOTOLOGY_USES_PYTHON=ON` I get two folders in `robotology-superbuild/build/install/lib/` :

```
Python3
└── dist-packages
    ├── yarp-3.10.1.dist-info
    │   ├── INSTALLER
    │   └── METADATA
    ├── yarp.py
    └── _yarp.so
```

```
Python3.12
└── dist-packages
    ├── icub_models
    │   ├── icub_models.py
    │   └── __init__.py
    ├── icub_models-3.0.0.dist-info
    │   ├── INSTALLER
    │   └── METADATA
    ├── icub.py
    └── _icub_python.so
```

When I ask Python to import yarp I get `No module named 'yarp'`

______________________

A dirty workaround is to simply copy the content of `/Python3` folder into `/Python3.12`  but ...
What is the best way to handle this problem ?

Many thanks in advance !

traversaro, 2024-12-09:
Thanks @MissingSignal for reporting the problem! 

Can you please specify:
* Which Linux distribution are you using?
* How did you installed the robotology-superbuild dependencies (i.e. `apt` or `conda`).
* Did you run the installation in a virtual environments? Did you installed any package via `pip` ?

To quickly report on the installed software, you can also run `curl -fsSL https://raw.githubusercontent.com/ami-iit/log-installed-software/main/src/log.py | python` and report the generated `installed_software.txt` file.

traversaro, 2024-12-09:
Also, can you report the content of `robotology-superbuild/build/install/share/robotology-superbuild/setup.sh` ?

traversaro, 2024-12-09:
I will also cherry-pick the fix on the `releases/2024.11` branch, so it is ready for a possible future `2024.11.2` release.

traversaro, 2024-12-09:
@MissingSignal the problem should have been fixed by https://github.com/robotology/robotology-superbuild/pull/1763, feel free to comment if you still have the problem.

closed 2024-12-09 by traversaro

MissingSignal, 2024-12-10:
@traversaro I confirm https://github.com/robotology/robotology-superbuild/pull/1763 fixed the problem.
Thank you for the  lightning-fast support 🙏🏻