robotology/robotology-superbuild #1798: Unstable branches do not use latest YARP February 2025

traversaro, 2025-02-18:
Since https://github.com/robotology/robotology-superbuild/pull/1789 and https://github.com/robotology/robotology-superbuild/issues/1788, the `Unstable` branches it is not using the latest YARP mainly for two reasons:
* https://github.com/robotology/yarp/pull/3169 introduced breaking changes in the transform interfaces, but at the moment there is not way to detect if a version of YARP has the new interface (a method was proposed in https://github.com/robotology/yarp/pull/3173 but not merged). So we can't easily modify our software to work with both the old and new interfaces
* Since https://github.com/robotology/yarp/pull/3171 YARP forces all downstream projects to use C++20 . Unfortunately this prevents the use of YARP with recent protobuf version compiled with c++17 that use abseil in their public headers, as abseil changes its ABI depending on the C++ version used (see https://github.com/abseil/abseil-cpp/issues/1280 and https://github.com/abseil/abseil-cpp/blob/master/FAQ.md#what-is-abi-and-why-dont-you-recommend-using-a-pre-compiled-version-of-abseil)

This details were already documented in other issues and PRs, but this issue is used to track the problem more at the high level.

randaz81, 2025-02-18:
Update: yarp 3.11 has been released

traversaro, 2025-03-08:
Fixed in https://github.com/robotology/robotology-superbuild/pull/1807 .

closed 2025-03-08 by traversaro