robotology/robotology-superbuild #1829: Report error while compiling the robotology-superbuild

Gianlucamilani, 2025-03-26:
- Yesterday together with @GiulioRomualdi we were trying to compile the robotology by typing:

`cmake --build . --config Release`

from the `build` folder : `(robenv2) C:\Users\gimilani\Documents\Github\robotology-superbuild\build>

Here the error message @traversaro 
`

```
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\basestate\gazebo_yarp_basestate.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\GazeboYAR
PPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/camera/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\camera\gazebo_yarp_camera.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\GazeboYARPPlugi
ns.vcxproj]
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\thrift\clock\gazebo_yarp_rpc_clock.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\GazeboYARPPlug
ins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/configurationoverride/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\configurationoverride\gazebo_yarp_configurationoverride.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-su
perbuild\build\GazeboYARPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/contactloadcellarray/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\contactloadcellarray\gazebo_yarp_contactloadcellarray.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-supe
rbuild\build\GazeboYARPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/controlboard/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\controlboard\gazebo_yarp_controlboard.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\Gaz
eboYARPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/depthCamera/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\depthCamera\gazebo_yarp_depthCamera.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\Gazeb
oYARPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/inertialmtbPart/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\inertialmtbPart\gazebo_yarp_distributedinertials.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuil
d\build\GazeboYARPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/externalwrench/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\externalwrench\gazebo_yarp_externalwrench.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build
\GazeboYARPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/fakecontrolboard/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\fakecontrolboard\gazebo_yarp_fakecontrolboard.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\b
uild\GazeboYARPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/forcetorque/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\forcetorque\gazebo_yarp_forcetorque.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\Gazeb
oYARPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/imu/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\imu\gazebo_yarp_imu.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\GazeboYARPPlugins.vcx
proj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/inertialmtb/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\inertialmtb\gazebo_yarp_inertial.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\GazeboYA
RPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/lasersensor/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\lasersensor\gazebo_yarp_lasersensor.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\Gazeb
oYARPPlugins.vcxproj]
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\thrift\linkattacher\gazebo_yarp_rpc_linkattacher.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\
GazeboYARPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/maissensor/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\maissensor\gazebo_yarp_maissensor.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\GazeboY
ARPPlugins.vcxproj]
[... 3 lines cut ...]
ld\build\GazeboYARPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/multicamera/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\multicamera\gazebo_yarp_multicamera.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\Gazeb
oYARPPlugins.vcxproj]
    Building Custom Rule C:/Users/gimilani/Documents/Github/robotology-superbuild/src/GazeboYARPPlugins/plugins/robotinterface/CMakeLists.txt
    Auto build dll exports
LINK : fatal error LNK1181: cannot open input file 'opencv_signal.lib' [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\src\GazeboYARPPlugins\plugins\robotinterface\gazebo_yarp_robotinterface.vcxproj] [C:\Users\gimilani\Documents\Github\robotology-superbuild\build
\GazeboYARPPlugins.vcxproj]
    Auto build dll exports
[... cut ...] 1847432e81\GazeboYARPPlugins-instal
l.rule;C:\Users\gimilani\Documents\Github\robotology-superbuild\build\CMakeFiles\08dbb4df0c6e47f40ff3ba7cf551d0db\GazeboYARPPlugins-complete.rule;C:\Users\gimilani\Documents\Github\robotology-superbuild\build\CMakeFiles\1fe77f4bdd06e1944d0b1163c6742108\GazeboYARPPlugins.rule;C:\
Users\gimilani\Documents\Github\robotology-superbuild\CMakeLists.txt' exited with code 1. [C:\Users\gimilani\Documents\Github\robotology-superbuild\build\GazeboYARPPlugins.vcxproj]
```

traversaro, 2025-03-26:
Can you report the `conda list` in the environment in which you are experiencing this issue?

traversaro, 2025-03-26:
Is this happening if you create a fresh environment? From the error message, I suspect somehow your conda environment was corrupted and some required files were deleted. However, if you quickly need to get it to work you can se the `ROBOTOLOGY_USES_GAZEBO` option to `OFF` to disable the compilation of the problematic repo.

Gianlucamilani, 2025-03-26:
```
# Name                    Version                   Build  Channel
ace                       8.0.0                he0c23c2_1    conda-forge
aiohttp                   3.9.5           py312he70551f_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.9.1                he0c23c2_0    conda-forge
asio                      1.29.0               h63175ca_0    conda-forge
assimp                    5.3.1                h0dbab56_3    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.20               h6823eb1_0    conda-forge
aws-c-cal                 0.6.12               hc83774a_0    conda-forge
aws-c-common              0.9.17               h2466b09_0    conda-forge
aws-c-compression         0.2.18               hc83774a_4    conda-forge
aws-c-event-stream        0.4.2               hc6c0aac_10    conda-forge
aws-c-http                0.8.1               hced5053_13    conda-forge
aws-c-io                  0.14.8               hebaacdb_0    conda-forge
aws-c-mqtt                0.10.4               hdafd9a4_2    conda-forge
aws-c-s3                  0.5.9                h7a83f0e_0    conda-forge
aws-c-sdkutils            0.1.16               hc83774a_0    conda-forge
aws-checksums             0.1.18               hc83774a_4    conda-forge
aws-crt-cpp               0.26.8              h672a689_11    conda-forge
aws-sdk-cpp               1.11.267             h12f3f85_8    conda-forge
azure-core-cpp            1.11.1               h249a519_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h91493d7_1    conda-forge
azure-storage-common-cpp  12.5.0               h91493d7_4    conda-forge
blosc                     1.21.6               h85f69ea_0    conda-forge
boost                     1.84.0               h7e22eef_3    conda-forge
brotli                    1.1.0                h2466b09_2    conda-forge
brotli-bin                1.1.0                h2466b09_2    conda-forge
bullet-cpp                3.25                 h2ab9e98_2    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
c-ares                    1.28.1               hcfcfb64_0    conda-forge
c-compiler                1.7.0                hcfcfb64_1    conda-forge
ca-certificates           2025.1.31            h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h91e5215_2    conda-forge
certifi                   2025.1.31          pyhd8ed1ab_0    conda-forge
cfitsio                   4.4.0                h9b0cee5_1    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.29.6               h400e5d1_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compilers                 1.7.0                h57928b3_1    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.3.0           py312hd5eb7cc_2    conda-forge
cppzmq                    4.10.0               h42135b4_1    conda-forge
cudatoolkit               11.8.0              h09e9e62_13    conda-forge
cxx-compiler              1.7.0                h91493d7_1    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dartsim                   6.13.2               h15b4ae7_2    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.4.1                h63175ca_0    conda-forge
double-conversion         3.3.0                h63175ca_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
exceptiongroup            1.2.2              pyhd8ed1ab_0    conda-forge
executing                 2.1.0              pyhd8ed1ab_0    conda-forge
expat                     2.6.2                h63175ca_0    conda-forge
fcl                       0.7.0                he22821c_4    conda-forge
ffmpeg                    6.1.1           gpl_h52c3a39_114    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
flann                     1.9.2                h23e6bae_0    conda-forge
fmt                       10.2.1               h181d51b_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_2    conda-forge
[... 275 lines cut ...]
win32_setctime            1.1.0              pyhd8ed1ab_0    conda-forge
wslink                    2.1.1              pyhd8ed1ab_0    conda-forge
x264                      1!164.3095           h8ffe710_2    conda-forge
x265                      3.5                  h2d74725_3    conda-forge
xerces-c                  3.2.5                h63175ca_0    conda-forge
[... cut ...]
yaml                      0.2.5                h8ffe710_2    conda-forge
yarl                      1.9.4           py312he70551f_0    conda-forge
zenoh-rust-abi            1.3.0.1.85.0         h1107ac9_0    conda-forge
zeromq                    4.3.5                he1f189c_4    conda-forge
zlib                      1.3.1                h2466b09_1    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge
zziplib                   0.13.69              h3ca93ac_2    conda-forge
```

Gianlucamilani, 2025-03-26:
[image]

do I need to set the cmake install prefix to `robotology-superbuild/build/install` as usual ?

Gianlucamilani, 2025-03-26:
I removed `build/src/GazeboYARPPlugins` .

I set the cmake install prefix to `robotology-superbuild/build/install`

Then I tried to recompile by typing:

`cmake --build . --config Release`

Now I get this error:

```
  The system cannot find the batch label specified - VCEnd
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(237,5): e
rror MSB8066: Custom build for 'C:\Users\gimilani\Documents\Github\robotology-superbuild\build\CMakeFiles\7c3fa7ff7b75e
38d764d0aede1338121\CppAD-update_disconnected.rule' exited with code 1. [C:\Users\gimilani\Documents\Github\robotology-
superbuild\build\CppAD.vcxproj]
```

Gianlucamilani, 2025-03-27:
I observe again the same error:

```
  The system cannot find the batch label specified - VCEnd
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(237,5): e
rror MSB8066: Custom build for 'C:\Users\gimilani\Documents\Github\robotology-superbuild\build\CMakeFiles\7c3fa7ff7b75e
38d764d0aede1338121\CppAD-update_disconnected.rule' exited with code 1. [C:\Users\gimilani\Documents\Github\robotology-
superbuild\build\CppAD.vcxproj]

```

cc @traversaro

Gianlucamilani, 2025-03-28:
> either build a new superbuild from scratch, or install the dependencies he need from conda binaries except for the projects we needs to modify (biomechanical-analysis-framework).

thanks a lot @traversaro for the support. Finally, I was able to install everything using the second option (installation via `conda`). Now the demo correctly runs on my Windows laptop with the updated version from `master`

closed 2025-03-28 by Gianlucamilani