robotology/robotology-superbuild #1831: Great CMake 4.0.0 failure of end of March 2025

traversaro, 2025-03-28:
In the last few days, there were a bit of (concurrent) changes on the conda-forge side:
* CMake 4.0.0 was released (https://github.com/conda-forge/cmake-feedstock/pull/236). This is the first CMake release with breaking changes in 11 years. 
* The `vs2019_win-64` (implicitly installed by the `cxx-compiler` metapackage) was released, with some regressions (see https://github.com/conda-forge/vc-feedstock/pull/93)

It is not clear how this two interacted (or if they interacted at all) but we are seeing some downstream CI failures in non-locked CI jobs (see https://github.com/conda-forge/vc-feedstock/issues/94#issuecomment-2760513869). 

This issue track the investigation and resolution of this problems. The nice news is that if you are using a locked environment, everything works fine. 

**TL;DR: If you are not using a working locked environment and you are experiencing problems, locking `vs2019_win-64=*=*_24` and `cmake<4.0.0` should be a sufficient workaround.**

h-vetinari, 2025-03-28:
> ideally soonish we should migrate to osqp 1.0.0

help in https://github.com/conda-forge/libosqp-feedstock/pull/13 would be welcome... 🙃

traversaro, 2025-03-28:
> > ideally soonish we should migrate to osqp 1.0.0
> 
> help in [conda-forge/libosqp-feedstock#13](https://github.com/conda-forge/libosqp-feedstock/pull/13) would be welcome... 🙃

Apologies, that and the protobuf migration are in my pipeline but this week has been a bit an infinite list of preemption by higher priorities. Indeed, here we do not use the osqp conda-forge package, but I would like to update to 1.0.0 here only once the conda-forge libosqp 1.0.0 is ready.

traversaro, 2025-03-28:
Following failure:

~~~
2025-03-28T10:31:04.2944536Z Not searching for unused variables given on the command line.
2025-03-28T10:31:04.2945767Z loading initial cache file /home/runner/work/robotology-superbuild/robotology-superbuild/b/src/OpenVR/CMakeFiles/YCMTmp/OpenVR-cache-Release.cmake
2025-03-28T10:31:04.2946958Z CMake Error at CMakeLists.txt:2 (cmake_minimum_required):
2025-03-28T10:31:04.2947642Z   Compatibility with CMake < 3.5 has been removed from CMake.
2025-03-28T10:31:04.2948141Z 
2025-03-28T10:31:04.2948525Z   Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
2025-03-28T10:31:04.2949728Z   to tell CMake that the project requires at least <min> but has been updated
2025-03-28T10:31:04.2950830Z   to work with policies introduced by <max> or earlier.
2025-03-28T10:31:04.2951416Z 
2025-03-28T10:31:04.2951878Z   Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.
2025-03-28T10:31:04.2952504Z 
2025-03-28T10:31:04.2952653Z 
2025-03-28T10:31:04.2952996Z -- Configuring incomplete, errors occurred!
~~~

As the plan is eventually to remove OpenVR, I guess just passing `DCMAKE_POLICY_VERSION_MINIMUM=3.10` is perfectly fine.

traversaro, 2025-03-28:
For unicycle planner instead it make sense to fix it upstream: https://github.com/robotology/unicycle-footstep-planner/pull/72 .

traversaro, 2025-03-29:
I still need to understand if this is related to CMake 4.0.0 or not, but I noticed that in local debugging in WSL, the configuration of YARP takes forever (but literally forever, several minutes and I had to stop it) on WSL + CMake 4.0.0 . When inspecting what the binary was doing with `strace`, it was continuously calling `stat` on all files in ``: 

~~~
stat("/mnt/c/Windows/system32/ddrawex.dll", {st_mode=S_IFREG|0555, st_size=77824, ...}) = 0
stat("/mnt/c/Windows/system32/de-DE", {st_mode=S_IFDIR|0555, st_size=4096, ...}) = 0
stat("/mnt/c/Windows/system32/declaredconfiguration.dll", {st_mode=S_IFREG|0555, st_size=36864, ...}) = 0
stat("/mnt/c/Windows/system32/DefaultAccountTile.png", {st_mode=S_IFREG|0555, st_size=4802, ...}) = 0
stat("/mnt/c/Windows/system32/DefaultDeviceManager.dll", {st_mode=S_IFREG|0555, st_size=42720, ...}) = 0
stat("/mnt/c/Windows/system32/DefaultHrtfs.bin", {st_mode=S_IFREG|0555, st_size=4227116, ...}) = 0
stat("/mnt/c/Windows/system32/DefaultPrinterProvider.dll", {st_mode=S_IFREG|0555, st_size=49152, ...}) = 0
stat("/mnt/c/Windows/system32/DefaultQuestions.json", {st_mode=S_IFREG|0555, st_size=858, ...}) = 0
stat("/mnt/c/Windows/system32/Defrag.exe", {st_mode=S_IFREG|0555, st_size=245760, ...}) = 0
stat("/mnt/c/Windows/system32/defragproxy.dll", {st_mode=S_IFREG|0555, st_size=36864, ...}) = 0
stat("/mnt/c/Windows/system32/defragres.dll", {st_mode=S_IFREG|0555, st_size=12288, ...}) = 0
stat("/mnt/c/Windows/system32/defragsvc.dll", {st_mode=S_IFREG|0555, st_size=552960, ...}) = 0
stat("/mnt/c/Windows/system32/delegatorprovider.dll", {st_mode=S_IFREG|0555, st_size=53248, ...}) = 0
stat("/mnt/c/Windows/system32/DeliveryOptimizationMIProv.mof", {st_mode=S_IFREG|0555, st_size=20268, ...}) = 0
stat("/mnt/c/Windows/system32/DeliveryOptimizationMIProvUninstall.mof", {st_mode=S_IFREG|0555, st_size=2044, ...}) = 0
stat("/mnt/c/Windows/system32/deploymentcsphelper.exe", {st_mode=S_IFREG|0555, st_size=57344, ...}) = 0
stat("/mnt/c/Windows/system32/deploymentcsps.dll", {st_mode=S_IFREG|0555, st_size=94208, ...}) = 0
stat("/mnt/c/Windows/system32/desk.cpl", {st_mode=S_IFREG|0555, st_size=237568, ...}) = 0
stat("/mnt/c/Windows/system32/deskadp.dll", {st_mode=S_IFREG|0555, st_size=73728, ...}) = 0
stat("/mnt/c/Windows/system32/deskmon.dll", {st_mode=S_IFREG|0555, st_size=65536, ...}) = 0
stat("/mnt/c/Windows/system32/desktopimgdownldr.exe", {st_mode=S_IFREG|0555, st_size=176128, ...}) = 0
stat("/mnt/c/Windows/system32/DesktopShellAppStateContract.dll", {st_mode=S_IFREG|0555, st_size=69632, ...}) = 0
stat("/mnt/c/Windows/system32/DesktopShellExt.dll", {st_mode=S_IFREG|0555, st_size=409600, ...}) = 0
stat("/mnt/c/Windows/system32/DesktopSwitcherDataModel.dll", {st_mode=S_IFREG|0555, st_size=393216, ...}) = 0
stat("/mnt/c/Windows/system32/DesktopView.Internal.Broker.dll", {st_mode=S_IFREG|0555, st_size=176128, ...}) = 0
stat("/mnt/c/Windows/system32/DesktopView.Internal.Broker.ProxyStub.dll", {st_mode=S_IFREG|0555, st_size=32768, ...}) = 0
stat("/mnt/c/Windows/system32/DetailedReading-Default.xml", {st_mode=S_IFREG|0555, st_size=3944, ...}) = 0
stat("/mnt/c/Windows/system32/DevDispItemProvider.dll", {st_mode=S_IFREG|0555, st_size=146040, ...}) = 0
stat("/mnt/c/Windows/system32/DeveloperOptionsSettingsHandlers.dll", {st_mode=S_IFREG|0555, st_size=389120, .

traversaro, 2025-03-29:
> I still need to understand if this is related to CMake 4.0.0 or not, but I noticed that in local debugging in WSL, the configuration of YARP takes forever (but literally forever, several minutes and I had to stop it) on WSL + CMake 4.0.0 . When inspecting what the binary was doing with `strace`, it was continuously calling `stat` on all files in `/mnt/c/Windows/system32/`:
> 
> ```
> stat("/mnt/c/Windows/system32/ddrawex.dll", {st_mode=S_IFREG|0555, st_size=77824, ...}) = 0
> stat("/mnt/c/Windows/system32/de-DE", {st_mode=S_IFDIR|0555, st_size=4096, ...}) = 0
> stat("/mnt/c/Windows/system32/declaredconfiguration.dll", {st_mode=S_IFREG|0555, st_size=36864, ...}) = 0
> stat("/mnt/c/Windows/system32/DefaultAccountTile.png", {st_mode=S_IFREG|0555, st_size=4802, ...}) = 0
> stat("/mnt/c/Windows/system32/DefaultDeviceManager.dll", {st_mode=S_IFREG|0555, st_size=42720, ...}) = 0
> stat("/mnt/c/Windows/system32/DefaultHrtfs.bin", {st_mode=S_IFREG|0555, st_size=4227116, ...}) = 0
> stat("/mnt/c/Windows/system32/DefaultPrinterProvider.dll", {st_mode=S_IFREG|0555, st_size=49152, ...}) = 0
> stat("/mnt/c/Windows/system32/DefaultQuestions.json", {st_mode=S_IFREG|0555, st_size=858, ...}) = 0
> stat("/mnt/c/Windows/system32/Defrag.exe", {st_mode=S_IFREG|0555, st_size=245760, ...}) = 0
> stat("/mnt/c/Windows/system32/defragproxy.dll", {st_mode=S_IFREG|0555, st_size=36864, ...}) = 0
> stat("/mnt/c/Windows/system32/defragres.dll", {st_mode=S_IFREG|0555, st_size=12288, ...}) = 0
> stat("/mnt/c/Windows/system32/defragsvc.dll", {st_mode=S_IFREG|0555, st_size=552960, ...}) = 0
> stat("/mnt/c/Windows/system32/delegatorprovider.dll", {st_mode=S_IFREG|0555, st_size=53248, ...}) = 0
> stat("/mnt/c/Windows/system32/DeliveryOptimizationMIProv.mof", {st_mode=S_IFREG|0555, st_size=20268, ...}) = 0
> stat("/mnt/c/Windows/system32/DeliveryOptimizationMIProvUninstall.mof", {st_mode=S_IFREG|0555, st_size=2044, ...}) = 0
> stat("/mnt/c/Windows/system32/deploymentcsphelper.exe", {st_mode=S_IFREG|0555, st_size=57344, ...}) = 0
> stat("/mnt/c/Windows/system32/deploymentcsps.dll", {st_mode=S_IFREG|0555, st_size=94208, ...}) = 0
> stat("/mnt/c/Windows/system32/desk.cpl", {st_mode=S_IFREG|0555, st_size=237568, ...}) = 0
> stat("/mnt/c/Windows/system32/deskadp.dll", {st_mode=S_IFREG|0555, st_size=73728, ...}) = 0
> stat("/mnt/c/Windows/system32/deskmon.dll", {st_mode=S_IFREG|0555, st_size=65536, ...}) = 0
> stat("/mnt/c/Windows/system32/desktopimgdownldr.exe", {st_mode=S_IFREG|0555, st_size=176128, ...}) = 0
> stat("/mnt/c/Windows/system32/DesktopShellAppStateContract.dll", {st_mode=S_IFREG|0555, st_size=69632, ...}) = 0
> stat("/mnt/c/Windows/system32/DesktopShellExt.dll", {st_mode=S_IFREG|0555, st_size=409600, ...}) = 0
> stat("/mnt/c/Windows/system32/DesktopSwitcherDataModel.dll", {st_mode=S_IFREG|0555, st_size=393216, ...}) = 0
> stat("/mnt/c/Windows/system32/DesktopView.Internal.Broker.dll", {st_mode=S_IFREG|0555, st_size=176128, ...}) = 0
> stat("/mnt/c/Windows/system32/DesktopView.Internal.Broker.ProxyStub.dll", {st_mode=S_IFREG|0555, st_size=32768, ...}) = 0
> stat("/mnt/c/Windows/system32/DetailedReading-Default.xml", {st_mode=S_IFREG|0555, st_size=3944, ...}) = 0
> stat("/mnt/c/Windows/system32/DevDispItemProvider.dll", {st_mode=S_IFREG|0555, st_size=146040, ...}) = 0
> stat("/mnt/c/Windows/system32/DeveloperOptionsSettingsHandlers.dll", {st_mode=S_IFREG|0555, st_size=389120, .
> ```
> 
> (this is just a small excerpt). Removing the `/mnt/c/Windows/system32/` from the `PATH` solved the problem, but as adding `/mnt/c/Windows/system32/` to the `PATH` is the default behaviour of WSL, this is something we need to investigate.

This is https://gitlab.kitware.com/cmake/cmake/-/issues/26817 . The nice thing is that the error goes away if one follows this section of the robotology-superbuild docs https://github.com/robotology/robotology-superbuild/?tab=readme-ov-file#sanitize-path-enviroment-variable-for-wsl .

closed 2025-03-29 by traversaro