robotology/robotology-superbuild #1854: Upgrade to YARP 3.12.0

traversaro, 2025-05-23:
YARP 3.12.0 will be released in a few days by branching out of the YARP's `master` branch. 

Unfortunately, I forgot that in https://github.com/robotology/robotology-superbuild/pull/1807 I had pinned the YARP version in `Unstable` branches to 3.11.1, so a whole ~3 months went on without the CI checking for CMake and/or C++ breaking changes in YARP, so we need do to that as soon as possible.

Beside C++ or CMake breaking changes, that we can easily handle with conditional on YARP version, there was also a rename of the (widely used) device `opencv_grabber` to `openCVgrabber` in https://github.com/robotology/yarp/pull/3225 . 
The `opencv_grabber` is widely used, at least in the AMI team, whenever a webcam is used to save data (typically as part of a setup used to save the video of an experiment as an external view, see https://github.com/search?q=org%3Aami-iit+opencv_grabber&type=code&p=1).

The rename of a YARP device is tricky as:
* It typically only catched at run time, so we don't catch regressions in CI
* The device name is typically embedded in a .xml yarprobotinterface file, so it is not easy to change the name of the device name to use based on the YARP version used

Possible way of dealing with that include:
* **SOL1**: We wait for the release for YARP 3.12.0, we update all the projects (public and private) that use `opencv_grabber` to `openCVgrabber`, and then we tag all of them, so we bump their version in LatestReleases all together. I am not a big fan of this, as them many user may want to use new projects with older YARP version (for example, when dealing with a robot setup that is shared and they do not control), so I typically prefer for project to support multiple YARP versions at the same time
* **SOL2**: We provide somehow in the superbuild a `opencv_grabber` device that just redirects to `openCVGrabber` printing a warning, and then we migrate in some time to `openCVGrabber` when we are sure that all ergoCub robots have been migrated to `openCVGrabber`

I need to investigate the feasibility of **SOL2**, but if that is technically possible I would probably go for that solution.

fyi @S-Dafarra @giotherobot @isorrentino @carloscp3009 @GiulioRomualdi @Giulero @HosameldinMohamed @xela-95 @LoreMoretti (tagging a few person across AMI team that I know that they use logger setups that use `opencv_grabber`).

S-Dafarra, 2025-05-23:
I think that mostly this affects the BLF logger files. I believe this should not be too hard to handle since the logger files are robot dependent. So I guess that if we keep the robot and the logger files aligned we should be fine

traversaro, 2025-05-23:
> I think that mostly this affects the BLF logger files. I believe this should not be too hard to handle since the logger files are robot dependent. So I guess that if we keep the robot and the logger files aligned we should be fine

Great, is this the same also for iRonCub @HosameldinMohamed @Giulero ? If yes, we can avoid to actively deal with the problem.

traversaro, 2025-05-23:
> I need to investigate the feasibility of **SOL2**, but if that is technically possible I would probably go for that solution.

@S-Dafarra comment actually subsumed this point, but just for curiosity: I discussed with a friend of mine and it seems that the solution may be feasible: https://chatgpt.com/share/683045ff-e598-8006-9f8b-4653050cd121 . If we ever want to do something like that, the only problem is: where to put this code? It will have some initial friction to add a new repo, but at some point I am really tempted to just have a a `yarp-backcompat-shim` repo to contain this back-compatibility logic (and also to keep other backcompat logic that is currently spread in different repos, like https://github.com/robotology/gz-sim-yarp-plugins/blob/719b105e7b5e2d26688436765ae578ac8ab06883/libraries/common/YarpDevReturnValueCompat.h#L4 and ).

traversaro, 2025-05-23:
Ok, a few error:

`ergocub-software` : 

~~~
2025-05-23T11:43:06.4755284Z /__w/robotology-superbuild/robotology-superbuild/build/src/ergocub-software/src/modules/couplingXCubHandMk5/yarp_plugin_couplingXCubHandMk5.cpp: In function 'void* couplingXCubHandMk5_create()':
2025-05-23T11:43:06.4757882Z /__w/robotology-superbuild/robotology-superbuild/build/src/ergocub-software/src/modules/couplingXCubHandMk5/yarp_plugin_couplingXCubHandMk5.cpp:30:50: error: invalid new-expression of abstract class type 'CouplingXCubHandMk5'
2025-05-23T11:43:06.4759958Z    30 | YARP_DEFINE_SHARED_SUBCLASS(couplingXCubHandMk5, CouplingXCubHandMk5, yarp::dev::DeviceDriver)
2025-05-23T11:43:06.4760711Z       |                                                  ^~~~~~~~~~~~~~~~~~~
2025-05-23T11:43:06.4762175Z In file included from /__w/robotology-superbuild/robotology-superbuild/build/src/ergocub-software/src/modules/couplingXCubHandMk5/yarp_plugin_couplingXCubHandMk5.cpp:10:
2025-05-23T11:43:06.4764953Z /__w/robotology-superbuild/robotology-superbuild/src/ergocub-software/src/modules/couplingXCubHandMk5/CouplingXCubHandMk5.h:41:7: note:   because the following virtual functions are pure within 'CouplingXCubHandMk5':
2025-05-23T11:43:06.4766617Z    41 | class CouplingXCubHandMk5 : public yarp::dev::DeviceDriver,
2025-05-23T11:43:06.4767150Z       |       ^~~~~~~~~~~~~~~~~~~
2025-05-23T11:43:06.4768295Z In file included from /__w/robotology-superbuild/robotology-superbuild/src/ergocub-software/src/modules/couplingXCubHandMk5/CouplingXCubHandMk5_ParamsParser.h:18,
2025-05-23T11:43:06.4770046Z                  from /__w/robotology-superbuild/robotology-superbuild/src/ergocub-software/src/modules/couplingXCubHandMk5/CouplingXCubHandMk5.h:15,
2025-05-23T11:43:06.4771783Z                  from /__w/robotology-superbuild/robotology-superbuild/build/src/ergocub-software/src/modules/couplingXCubHandMk5/yarp_plugin_couplingXCubHandMk5.cpp:10:
2025-05-23T11:43:06.4774121Z /__w/robotology-superbuild/robotology-superbuild/build/install/include/yarp/dev/IDeviceDriverParams.h:64:18: note:     'virtual bool yarp::dev::IDeviceDriverParams::getParamValue(const string&, std::string&) const'
2025-05-23T11:43:06.4775941Z    64 |     virtual bool getParamValue(const std::string& paramName, std::string& paramValue) const = 0;
2025-05-23T11:43:06.4776642Z       |                  ^~~~~~~~~~~~~
2025-05-23T11:43:06.4778014Z /__w/robotology-superbuild/robotology-superbuild/build/install/include/yarp/dev/IDeviceDriverParams.h:70:25: note:     'virtual std::string yarp::dev::IDeviceDriverParams::getConfiguration() const'
2025-05-23T11:43:06.4779429Z    70 |     virtual std::string getConfiguration() const = 0;
2025-05-23T11:43:06.4779912Z       |                         ^~~~~~~~~~~~~~~~
~~~

that sounds the same problem reported in https://github.com/robotology/yarp/pull/3225#issuecomment-2903743027 .

## `gazebo-yarp-plugins`

~~~
2025-05-23T11:44:01.9981643Z FAILED: plugins/camera/CMakeFiles/gazebo_yarp_camera.dir/src/CameraDriver.cpp.o 
2025-05-23T11:44:01.9994171Z /usr/bin/c++ -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DDEBUG=1 -DLIBBULLET_VERSION=3.05 -DLIBBULLET_VERSION_GT_282 -Dgazebo_yarp_camera_EXPORTS -I/__w/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/camera/include/gazebo -I/__w/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/camera/include/yarp/dev -I/__w/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/camera/include -I/__w/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/libraries/common/include -I/__w/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/libraries/singleton/include -isystem /usr/include/gazebo-11 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-9.7 -isystem /usr/include/ignition/math6 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport8 -isystem /usr/include/ignition/msgs5 -isystem /usr/include/ignition/common3 -isystem /usr/include/ignition/fuel_tools4 -isystem /__w/robotology-superbuild/robotology-superbuild/build/install/include -isystem /usr/include/eigen3 -isystem /usr/include/sdformat-9.7/sdf/.. -isystem /usr/include/ignition/cmake2 -isystem /usr/include/uuid -isystem /usr/include/opencv4 -g -Wall -Wextra -pedantic -Weffc++ -Woverloaded-virtual -fPIC -I/usr/include/uuid -MD -MT plugins/camera/CMakeFiles/gazebo_yarp_camera.dir/src/CameraDriver.cpp.o -MF plugins/camera/CMakeFiles/gazebo_yarp_camera.dir/src/CameraDriver.cpp.o.d -o plugins/camera/CMakeFiles/gazebo_yarp_camera.dir/src/CameraDriver.cpp.o -c /__w/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/camera/src/CameraDriver.cpp
2025-05-23T11:44:02.0006925Z In file included from /__w/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/camera/src/CameraDriver.cpp:8:
2025-05-23T11:44:02.0008930Z /__w/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/camera/include/yarp/dev/CameraDriver.h:11:10: fatal error: yarp/dev/FrameGrabberInterfaces.h: No such file or directory
2025-05-23T11:44:02.0010439Z    11 | #include <yarp/dev/FrameGrabberInterfaces.h>
[... cut ...]
2025-05-23T11:44:02.2191734Z [77/469] Building CXX object src/libraries/icubmod/cartesianController/CMakeFiles/cartesiancontrollerclient.dir/yarp_plugin_icubmod_cartesiancontrollerclient.cpp.o
2025-05-23T11:44:02.2193819Z [78/469] Building CXX object src/libraries/icubmod/cartesianController/CMakeFiles/cartesiancontrollerclient.dir/ClientCartesianController.cpp.o
2025-05-23T11:44:02.2195526Z [79/469] Building CXX object src/libraries/icubmod/cartesianController/CMakeFiles/cartesiancontrollerserver.dir/ServerCartesianController.cpp.o
2025-05-23T11:44:02.2196662Z ninja: build stopped: subcommand failed.
~~~

again just deprecated headers? To test.

traversaro, 2025-05-23:
> again just deprecated headers? See https://github.com/robotology/yarp/blob/8300e5242a335eab192a7d49801a46ed43593647/doc/release/v3_5_0.md?plain=1#L186 .

Interesting, it seems that no warning is printed for some reason.

traversaro, 2025-05-25:
Some related changes:
* https://github.com/ami-iit/bipedal-locomotion-framework/pull/970
* https://github.com/robotology/icub-main/pull/1021
* https://github.com/robotology/walking-teleoperation/pull/157
* https://github.com/robotology/icub-main/pull/1022
* https://github.com/icub-tech-iit/ergocub-software/pull/310

closed 2025-06-07 by traversaro