robotology/robotology-superbuild #1870: Missing ROS2 carriers

MissingSignal, 2025-06-12:
Dear all,

I've built from source the latest version of the superbuild with ROS2 support enabled.
When running the yarp server I get this error:

```
Call with --help for information on available options
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
[ERROR] **Missing one or more required carriers** 
[ERROR] for yarpserver --ros (rossrv, tcpros, xmlrpc).
[ERROR] Run 'yarp connect --list-carriers' to see carriers on your machine
[ERROR] Aborting.
```

when I list the available carriers I see that all the ros carriers ( rossrv etc ..) are missing.

_it is my first time compiling with ROS2 support, I might have missed something during the installation._

Many thanks !

Superbuild (rolling update/stable): YARP 3.12.0+1-20250604.3+git285f6317e
ROS2: Jazzy
OS: Ubuntu 24.04

MissingSignal, 2025-06-12:
Thank You @traversaro , I was aware of this.
I'm trying to use YARP + ROS2 , is it possible ?

traversaro, 2025-06-12:
> Thank You @traversaro , I was aware of this.
> I'm trying to use YARP + ROS2 , is it possible ?
> 

Yes, but there is no direct equivalent of the ROS 1 support. The support for ROS 2 is in the form of NWS and NWC that public data from YARP interfaces to ROS 2 topics. If you explain your use case we can probably provide more specific details.

MissingSignal, 2025-06-12:
Thank you, your reply clarifies my problem. 
Having seen the -ros option when launching the yarp server (and in the yarpmanager), I was expecting it to work with ros2.

I was looking for a way to expose the ports of my module as ros topics (and vice versa).

traversaro, 2025-06-12:
> Thank you, your reply clarifies my problem. 
> Having seen the -ros option when launching the yarp server (and in the yarpmanager), I was expecting it to work with ros2.
> 
> I was looking for a way to expose the ports of my module as ros topics (and vice versa).
> 

In that case either a bridge or directly publishing the ROS 2 topic from your module I guess are the easiest option.

MissingSignal, 2025-06-12:
I'll do as you suggest.
Thank you very much @traversaro 🙏