robotology/robotology-superbuild #1881: `icub-firmware-shared` downloaded with a different folder name

isorrentino, 2025-07-16:
When enabling the ROBOTOLOGY_ENABLE_ICUB_HEAD option, the `icub-firmware-shared` repository is downloaded with the name `icub_firmware_shared` instead of icub-firmware-shared as expected. This discrepancy in naming conventions between the repository name (`icub_firmware_shared`) and its intended name (`icub-firmware-shared`) may cause confusion as it happened to me today.

cc @traversaro

Nicogene, 2025-07-17:
This is probably due to the name of the project is `icub_firmware_shared`:
- https://github.com/robotology/icub-firmware-shared/blob/376f47cddbc28be1a522c049308dfa36f09e1bf7/CMakeLists.txt#L8

isorrentino, 2025-07-17:
> Just to understand, is this just a problem of confusion or there is some tooling (for example in firmware building) somewhere that hardcode the repo name of `icub-firmware-shared` ?

The issue arose when I tried to build the EMS firmware and couldn't find the libraries that are contained in icub-firmware-shared. Indeed when we build the code, it expects to find `icub-firmware-shared` (see https://github.com/robotology/icub-firmware/blob/d8e5c72e7bf0e2d9091669517fac1c41f97fc030/emBODY/eBcode/arch-arm/board/ems004/env/eLoader/proj/eLoader.uvprojx#L342 for example).

traversaro, 2025-07-17:
https://github.com/robotology/ycm-cmake-modules/pull/480 and https://github.com/robotology/robotology-superbuild/pull/1882 should solve the issue. @isorrentino can you briefly check it (even without testing) so that I can merge and release? Thanks!

isorrentino, 2025-07-17:
I attempted to test https://github.com/robotology/ycm-cmake-modules/pull/480 with https://github.com/robotology/robotology-superbuild/pull/1882 but the repository is still downloaded with the name `icub_firmware_shared`

isorrentino, 2025-07-17:
After the merge, I tested it again and it works. Great, thanks @traversaro!